WEC-Sim/WEC-Sim #670: [Feature request] Memory usage - Flag to save (or not) panels pressure with non-linear hydrodynamics

mmartini1987, 2021-08-10:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
_Dear WEC-Sim developers, I am running a non-linear hydrodynamic simulation (nlHydro=2) of a flap-type device. I am unable to run simulations longer than approx. 3600s, as my computer runs out of memory. I noticed that in the workspace there are very large variables related to non-linear hydrodynamics that likely collapse my memory, namely: "body1_hspressure_out", "body1_wavelinearpressure_out", "body1_wavenonlinearpressure_out" and so on. I understand these are not required for the simulation to run, but only to store pressure values on each panel._

**Describe the solution you'd like**
_Is there a way to avoid these variables to be saved into the workspace? I could not find anything on the WEC-Sim guide._

**Describe alternatives you've considered**
_I considered modifying myself the WEC-Sim source code but I do not want to alter the code._

**Personal project context**
_N/A_

**Additional context**
_N/A_

akeeste, 2021-08-10:
Hello @mmartini1987 

Thanks for reopening with the template. Currently I do not see a way to use the nonlinear hydro option without saving the body pressure data. You can currently change the output time step for the pressure data with ``simu.dtNL`` but that also changes the time  step of the force calculations. I will link this issue to our project board for a feature enhancement. 

Can you also give the follow for context:
- Number of hydrodynamic bodies 
- Approximate number of stl panels for each body 
- simu.dt, simu.dtOut

I am a bit surprised that the number of panels does not cause an unusably slow computational speed before the computer memory is filled. Though I can see how this might happen for a very long computation time with a small time step.

Adam

akeeste, 2021-08-10:
@kmruehl Thanks for the clarification. 

@mmartini1987 Actually we do have a variable that should determine if pressure data is saved: ``simu.pressureDis``. However right now this only determines if the ``body1_hspressure_out`` and other pressure variables are saved into the response class, not whether Simulink saves the variable in the first place.

mmartini1987, 2021-08-11:
Thank you very much for your prompt and kind response - and apologies if I initially opened the issue without using the template (nice and simple way to organise the forum, by the way).

@kmruehl I have two hydrodynamic bodies for this device: a base, fixed to the bottom, with ~2600 STL panels, and a flap, hinged at the base, with ~8500 STL panels. I am counting STL panels by counting the occurrences of "facet normal" in the STL file, opening it with Notepad++. Such numbers are confirmed when I look at the variables "bodyX_wavelinearpressure_out" and similar. For reference, the GDF files have 668 and 2129 panels for the base and the flap, respectively. I note that the "flap.stl" file within the OSWEC example provided with WEC-Sim accounts for ~1000 panels. Would you suggest a lower number of STL panels? For information, I have created the STL files in Rhino from the GDF by clicking on File->Save As, then picking "Stereolithography (.stl)", then '"Save geometry only", then "ASCII" as file type. 

Additionally, simu.dt and simu.dtOut are 0.05 s; I have my simulation crashed when simu.endTime is 10900 s (I am at running 3 hours simulations, I have a ramp-up time of 100 s).

@akeeste, thanks for the detail, so I assume all pressure values for all panels are currently stored into the workspace during the simulation and then passed to the response class.

closed 2021-08-12 by mmartini1987