WEC-Sim/WEC-Sim #673: [Developer Issue] Github Actions support for MATLAB on Windows

H0R5E, 2021-08-11:
**Project**

WEC-Sim

**Describe the issue**

[Matlab-actions](https://github.com/matlab-actions/overview) doesn't yet support Windows installations of MATLAB, but they are working on it according to [this issue](https://github.com/matlab-actions/setup-matlab/issues/18).

This issue is for tracking support for Windows and updating the workflows when it is available.

H0R5E, 2022-02-04:
Windows and MacOS support [is now available](https://github.com/matlab-actions/setup-matlab/issues/18#issuecomment-1006990188),

H0R5E, 2022-02-24:
I have investigated adding Windows and MacOS runner, but unfortunately the Simscape toolboxes are not yet available, as per [this comment](https://github.com/matlab-actions/setup-matlab/issues/18#issuecomment-1006990188):

> matlab-actions/setup-matlab@v1.1.0 contains limited, undocumented, support for Windows and macOS GitHub-hosted runners. On these platforms, only MATLAB and Simulink are installed **(no toolboxes)** and only releases R2021a and above are available.

I'll keep checking for updates.

H0R5E, 2023-01-05:
Windows support is now in beta, so I will investigate supporting WEC-Sim.

H0R5E, 2023-01-12:
I have started a [feature branch](https://github.com/H0R5E/WEC-Sim/tree/github_actions_windows) on my fork to test this out. The tests on the Windows' runners aren't working at the moment. I'm having a discussion with the guys from Mathworks about it in [this thread](https://github.com/matlab-actions/setup-matlab/issues/18).

kmruehl, 2023-02-08:
@H0R5E any update on this?

H0R5E, 2023-02-09:
No movement on the thread since 3 weeks ago. I will ask if any progress has been made.

kmruehl, 2023-05-17:
I'm closing this issues since it's been moved to the [project board here](https://github.com/WEC-Sim/WEC-Sim/projects/58#card-89245605)

closed 2023-05-17 by kmruehl