WEC-Sim/WEC-Sim #674: [BUG] MoorDyn with Paraview - RM3 Application

akeeste, 2021-08-11:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
The Paraview application case using the RM3 with MoorDyn is not working. It seems the Lines.out file is not yet written completely when read, which causes issues when reading and manipulating the MoorDyn data in the post-processing stage.

**WEC-Sim file(s)**
write_paraview_mooring.m - line 31
responseClass.m - loadMoorDyn() method
Lines.out

**To Reproduce**
Steps to reproduce the behavior:
1. Install the latest version of MoorDyn for WEC-Sim at: https://github.com/WEC-Sim/MoorDyn. Run BEMIO.
2. Run the applications case: Paraview_Visualization\RM3_MoorDyn_Viz
3. See error:
```
Error using wecSim (line 45)
Error evaluating 'StopFcn' callback of Frames block (mask) 'RM3MoorDyn/Global Reference Frame'.
Callback string is 'run('wecSimStopFunction')'
Caused by:
    Error using wecSim (line 45)
    X and V must be of the same length.
```

Steps to better see the error during simulation:
1. Install the latest version of MoorDyn for WEC-Sim at: https://github.com/WEC-Sim/MoorDyn. Run BEMIO.
2. Place a breakpoint in responseClass.m loadMoorDyn() method (~line 228).
2. Run the applications case: Paraview_Visualization\RM3_MoorDyn_Viz
3. While the simulation is paused, open moordyn\Lines.out and view data.
    
When I stop the simulation at the breakpoint and reopen Lines.out, it appears fully written. Has anyone else seen this issue? Perhaps we can put in a check that ensures MoorDyn has fully written the output files before reading them back in.

**Expected behavior**
I expect the case to run as is without issue, and to fully print all output data to Lines.out

**Screenshots**
Error message:
[image]

Partially written Lines.out:
[image]

Fully written Lines.out after simulation:
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB/Simulink 2021a
 - WEC-Sim v4.3

**Additional context**
Discovered this while reviewing [Applications PR 7](https://github.com/WEC-Sim/WEC-Sim_Applications/pull/7). However the updates are not actually causing this issue (rather the PR is revealing the issue as it should).

H0R5E, 2021-08-12:
I can confirm this bug.

It seems that MoorDyn does not flush its output buffers before the `LinesCalc` function returns. When `LinesCalc` returns, MATLAB takes back control its thread before Moordyn can deallocate its objects which would then flush the buffers. This is why the MoorDyn windows stays open even though the call has finished.

I think the only way to fix this is to change the MoorDyn code, however it's not clear to me where the version of MoorDyn we are using is stored. This also seems to be in violation of the GPL  as there is no source code available in [WEC-Sim/MoorDyn](https://github.com/WEC-Sim/MoorDyn), but we are distributing compiled versions. I'd say we could link to [mattEhall/MoorDyn](https://github.com/mattEhall/MoorDyn) to fix it, but the code is not the same as in that repo, so it wouldn't be sufficient for compliance.

H0R5E, 2021-08-12:
An alternative option to changing the MoorDyn code might be to prepare the MATLAB interface [in the recommended manner](https://uk.mathworks.com/help/matlab/matlab_external/publish-interface-to-shared-c-library.html), if we're not doing so already. There is no guarantee this would fix the issue, though.

H0R5E, 2021-08-16:
While I was investigating #677 and #682, I noticed that the code for calling MoorDyn is in a couple of different places. There is code defined in the callbacks of the `WECSim_Lib/Moorings/MoorDyn/MoorDyn` block, while the function defined in `WECSim_Lib/Moorings/MoorDyn/MoorDyn/MoorDyn/Triggered Subsystem` calls the function through the `MoorDyn_caller` function in the `MoorDyn_caller.m` file. Why not just do all the work in one place? Is there a reason for it?

H0R5E, 2021-08-16:
I had a quick play with moving some of this code around, and it seemed to help close down MoorDyn after it had finished, but it would either result in an error at the end of the run or crash MATLAB. I guess there is a magic combination somewhere. I can keep trying if no one else has any better ideas.

akeeste, 2021-11-17:
This issue has been difficult to reproduce consistently. We are also looking to add MoorDyn v2 in the near future which might resolve the issue. Closing this issue for now due to these factors.

closed 2021-11-17 by akeeste

akeeste, 2021-11-17:
> Sorry, I dropped the ball on this one.

@H0R5E No worries. I couldn't find a way to cause this issue consistently so it dropped off my radar too.