WEC-Sim/WEC-Sim #680: [WEC-Sim Applications/examples/RM3_DD_PTO]

sfdrrsul, 2021-08-13:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Error: While running the example, RM3_DD_PTO ._

**Describe the bug or issue**
_Variant control 'sv_b_dragBody' of variant block 'RM3_DD_PTO_1/Float' must be a condition expression or name of a Simulink.Variant object in the global scope workspace of the model. Ensure that the condition does not match the model name_

**To Reproduce**
_https://github.com/WEC-Sim/WEC-Sim/issues/144  I followed the steps mentioned in this issue 144 but I am unable to find the solution_

**Expected behavior**
_It should run._

**Screenshots**
_[image]._
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [ Windows 10]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2020b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v4.3]

**Additional context**
_Thanking you in advance._

akeeste, 2021-08-13:
Hi @sfdrrsul 

Thanks for opening an issue using a template. I have tested this Application on the same workstation but unfortunately cannot recreate your issue. I recommend ensuring that you have installed the latest version of both WEC-Sim and the Applications repository, and that all library links are unbroken.

Have you made any changes to the library or simulink models after installation? Also, can you tell me what method you are using to run WEC-Sim? (i.e. command line, [as a function](http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#running-as-function), or [from simulink](http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#running-as-function))

sfdrrsul, 2021-08-16:
Hi akeeste,
Thank you for support.
I am using a latest version 4.3
[image]
I donot completely understand what you mean by the library links are unbroken, but i can see the link sign on the simulink block is appeaing. Which I assume it is linked.
[image]
 I did not changed the libray after the installation =.
 
 I am trying to run the using simulink, just double clicking the file availble in the directroty of MATLAB where the RM3 is avaible.
[image]
 The same error is appearing for RM3.slx as well.
[image]
 
 I have inseterd the screenshorts, so that it could be easier for you to understand the issue.
 
Again thank you for your time.

akeeste, 2021-08-16:
Hi @sfdrrsul,

Thank you for the extra information. I don't see an issue with your library or links. It seems this issue is related to how you are [running WEC-Sim from Simulink](http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#running-from-simulink). Currently, this cannot be a one-step method like the ``wecSim`` command. To run from Simulink you must:
1. Open the relevant Simulink model.
2. Type ``wecSimInitialize`` in the Command Window
3. Run the model from Simulink

When skipping Step 2, I see similar issues to what you are describing. If you open the model and use the ``wecSimInitialize`` command before running in Simulink, this should resolve your issue.

Please let me know if this solution does not work.

Adam

sfdrrsul, 2021-08-17:
Hi Adam,

I really appreciate your prompt response and support you are providing.

I tried the steps you mentioned, i.e. i) Open the relevant WEC-Sim Simulink file, ii) Type wecSimInitialize in the Command Window iii) Hit Play in Simulink model to run.
 but when I run the wecSimInitialize, the following error comes;
[image]

I think, I have installed the WEC-SIM correctly, that is why I am able to run the base examle with wecSim command. This can be seen in the attached screenshort.
[image]

I downloaded all the files from the github folder. 

I am not sure about the issue.
If you are able to get any clue from the error coming after running wecSimInitialize.
Once again thank you  very much for your support.
Regards
Safdar

akeeste, 2021-08-17:
Hi Safdar,

This error is occurring because the ``Global Reference Frame`` Parameter Input Method is set to ``Custom Parameters``. By default, the body mask lists the general format of a path to the hydrodata ("hydrodata/*.h5"), but not a specific .h5 file. In this case, you will have to complete the mask information in all blocks to use the correct values and call the correct files.

Alternatively, you can change the ``Global Reference Frame`` mask to use the ``Input File`` setting. In this case, the Simulink model will read in an input file and run as intended.

This default should be ``Input File``. Updating this default and adding to the documentation on running from Simulink is on our [project board ](https://github.com/WEC-Sim/WEC-Sim/projects/58#card-67002382) for the near future.

Adam

sfdrrsul, 2021-08-18:
Hi Adam,

Yes, it worked.
I really appreciate your efforts and support.

Safdar

closed 2021-08-18 by sfdrrsul