WEC-Sim/WEC-Sim #681: [BUG] BEMIO - Error using hdf5lib2

watchara3512, 2021-08-14:
**Describe the bug**

Error using hdf5lib2

**WEC-Sim file(s)**

For the Boundary Element Method Input/Output (BEMIO), I am trying to run it (refer to the Two-Body Point Absorber (RM3) tutorial & example-BEMIO-WAMIT-Cylinder) but I get the following errors as shown in Fig. 1 & 2. 
[image]
Fig. 1
[image]
Fig. 2

**To Reproduce**

I downloaded the HDF5 files from https://www.hdfgroup.org/downloads/hdf5/ (as shown in Fig. 3) and installed them. However, there is still an error about hdf5lib2 as shown in Fig. 1 & 2.
[image]
Fig. 3

**Expected behavior**

It should run.

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - MATLAB/Simulink Release [2019b]
 - WEC-Sim Release [v4.3]

**Additional context**
Thank you in advance

akeeste, 2021-08-16:
Hi @watchara3512 

Thank you for the additional information. I have not seen this error before and a solution does not come up in any closed issues. I tested these functions with the same workstation and did not have issues. I have never manually installed hdf5 into MATLAB, so this should not be necessary to use the ``Write_H5()`` BEMIO function. I will consult with the rest of the development team and get back to you. Unfortunately if this is a MATLAB issue outside of WEC-Sim, we are unable to provide support.

@rodrijardim Did you find a solution to this problem in issue #531?

Thanks,
Adam

watchara3512, 2021-08-16:
Hi @akeeste 

Thank you very much for clarifying and your support.

Watchara

closed 2021-08-21 by watchara3512