WEC-Sim/WEC-Sim #683: ERROR during import data from ANSYS AQWA

Dadidal, 2021-08-20:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
BEMIO import from Ansys Aqwa

**Describe the bug or issue**
I calculated the Hydrodynamic Response of structure using ANSYS Aqwa (Hydrodynamic Diffraction module in ANSYS Workbench). During the import of hydrodata from Ansys Aqwa and creating the .h5 file I got the following ERROR:

Unable to perform assignment because the size of the left side is 6-by-1 and the size of the right
side is 1-by-5.

Error in Read_AQWA (line 232)
                hydro(F).sc_ph((kdiff-1)*6+1:kdiff*6,j,i) = tmp2;  % phase of scattering force

**To Reproduce**
1. Create bemio.m file for import from Ansys Aqwa
2. Run bemio.m
3. See error

**Expected behavior**
Generated .h5 file

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB/Simulink 2021a
 - WEC-Sim Release v4.3
 - ANSYS 2021R1

**Additional context**
How I can solve this problem and import the hydrodata from .AH1 and .LIS files generated using Aqwa?

Dadidal, 2021-08-22:
[hydroData.zip](https://github.com/WEC-Sim/WEC-Sim/files/7027285/hydroData.zip)

dforbush2, 2021-08-23:
Hi @Dadidal 

We're not AQWA experts, so the support we can provide may not be comprehensive. 

AQWA (very unfortunately) changes the content and structure of their output files with every new edition. So we often encounter versioning issues, and this may be the case here since you are using a new edition, but I do not see anything that suggests this is the case. 

What I am seeing when I try to run your files is that, for some reason, the variable `raw2`  which should read in the .LIS file is only read in part of the way, and thus is unable to return meaningful results, resulting in the error you are seeing. This suggests that maybe the EOL characters in the textscan function are not, for some reason, working. Will follow up.

nathanmtom, 2021-09-10:
@Dadidal The WEC-Sim team wanted to check-in to see if @dforbush2 response above has answered your initial questions? 

Cheers, 
Nathan 
National Renewable Energy Laboratory

dforbush2, 2021-09-28:
@Dadidal 

massive apologies for the delay. This is an issue related to the version of AQWA. I don't know exactly when it started, but the fundamental problem is that old versions splits different headings into different tables: like this:
[image]
[image]

But in your example file: 
[image]

So the line and spacing count to read in this data is not working properly. Data is being read incorrectly even before the error triggers. Unfortunately, other than changing versions or tweaking AQWA parameters such that the output matches the former example, I do not have a good work-around for you. 

This issue will remain open until I push a fix to Read_AQWA to address this issue. I may ask you to test this fix with other files generated with your AQWA version.

Best,
Dom

dforbush2, 2021-09-30:
Upon further investigation this is not likely a version issue, but an AQWA configuration issue that dictates the organization of this table.

kmruehl, 2021-12-07:
@Dadidal this should be resolved with #761

closed 2021-12-07 by kmruehl