WEC-Sim/WEC-Sim #684: A problem of running simulation in WEC-SIM

Doudou-Li, 2021-08-21:
Hello,

I usually use your beautiful WEC-Sim (Wave Energy Converter SIMulator)  with no problem of running simulation on my computer.

Recently, I want to utilize it on my laptop ​for the first time.
​But  I find there is a  problem when I run the wecSim.m file which is shown as follow:

Error using wecSim (line 225)
'simu' is not allowed as the left-hand side of 'simu.yawNonLin' in variant  condition expression
'typeNum>=10 && typeNum<20 && simu.yawNonLin~=1'. Only enumerated type names are allowed.

I think I have built a startup.m file successfully on my laptop, copied the same WEC-Sim-master files, and have done the right procedures as I did on my computer.
So I am very confused why it shows an error using problem.

Could you help me to solve this problem.
Thanks a lot.
Doudou

dforbush2, 2021-08-23:
Hi Doudou,

Can you please resubmit this issue using one of our issue templates? It helps us more effectively address and reference these issues.

https://github.com/WEC-Sim/WEC-Sim/issues/new/choose

Thanks,
Dom

closed 2021-08-23 by dforbush2

Doudou-Li, 2021-08-23:
Hi, Dom

I found that it is a matlab verison problem.
I have changed my matlab 2017a version to a new matlab 2020a one on my laptop. 
And I could run the simulation in WEC-SIM successfully now.

Thank you for your reply
Doudou

dforbush2, 2021-08-23:
Thank you for response. Due to a number of new MATLAB features added in recent revisions, we have updated our supported versions to MATLAB 2019b. While most features will still work in older versions of MATLAB there are some compatibility issues.

http://wec-sim.github.io/WEC-Sim/master/user/getting_started.html#matlab-requirements

Doudou-Li, 2021-08-23:
Very good. Thanks

Doudou

http://wec-sim.github.io/WEC-Sim/master/user/getting_started.html#matlab-requirements