WEC-Sim/WEC-Sim #693: Response for Free Decay Tests for OSWECs with an initial angle

salhus, 2021-08-31:
**Project**
OSWEC

**Describe the issue**

When conducting free-decay tests for OSWEC type WECs, if I give an initial angular displacement using setInitDisp method, the pitch response starts from 0 and not from the initial angle set. On closer inspection, the pitch response has a bias, and the crest to trough difference corresponds to the initial angular displacement.
[image]
[image]
[image]
[wecsim_input_file.txt](https://github.com/WEC-Sim/WEC-Sim/files/7086066/wecsim_input_file.txt)

nathanmtom, 2021-09-07:
@salhus Based on our discussion last week, I just wanted to clarify that this bias is from the `output.plotResponse(1,5)` function, correct? You mentioned that when looking at the `output.bodies(1).position(1,5)` the initial displacement is accurate. If you can confirm then we can look at this function to see if it needs to be updated.

salhus, 2021-09-07:
@nathanmtom  Yes, the output class has the accurate values, but the plotResponse(1,5) method plots biased results

jtgrasb, 2021-09-08:
@salhus I am having trouble replicating the issue. What exactly did you put in the input file to produce the initial rotation?

salhus, 2021-09-08:
Hi @jtgrasb  Sorry my bad,
here is the code for initial angular position I used,
intAng = deg2rad( #your angle here in deg  );
body(1).setInitDisp([0,0,-0.2352]-[0,0,-0.5], [0 1 0 intAng], [0 0 0]);

jtgrasb, 2021-09-10:
@salhus Thanks for the response. I was able to replicate your issue and find the issue within the plotResponse function. The function is subtracting the initial position from the position throughout the motion, producing the bias. 

pos=obj.bodies(bodyNum).position(:,comp) - obj.bodies(bodyNum).position(1,comp);

This can be easily changed as shown below to just plot the position with no bias, and the plot is shown correctly below.

pos=obj.bodies(bodyNum).position(:,comp); 

I am not sure why the initial position is automatically subtracted from each position. It seems like a strange concept to include without a specific reason. I think we should discuss this next meeting before fixing it as there may be a reason it was done this way.

[image]

[image]

salhus, 2021-09-10:
@jtgrasb  Thanks for spotting that. Yes that sounds good.  
Maybe it's needed for some special case. But yes we should discuss this

nathanmtom, 2021-09-10:
@jtgrasb @salhus Thanks for the updates to this discussion. I wanted to chime in that the reason that the function likely was coded to subtract off the initial position is because for the translational modes the output will be where the center of gravity is located. So if the vertical center of gravity is a -1 m then at t=0 s the output would start at -1 m which would make it seem like the device has been submerged from the device's equilibrium condition. By default there is no angle related to the COG so I do not see any reason to subtract off the initial rotational angles. 

It may be easiest to set up an if statement such that if `comp` is 1, 2, or 3 to subtract off the locations of the center of gravity while not subtracting any values for the rotational modes 4, 5, 6. If we stick with taking the response at t=0 this will effect any free decay tests by starting at 0.

salhus, 2021-09-10:
@nathanmtom @jtgrasb  Thanks!
I suppose a PR could be made to resolve this.
Let me know if @jtgrasb  you would like to do that since you spotted this. Or I could do it too.

jtgrasb, 2021-09-10:
@nathanmtom @salhus That makes sense, thanks! I am getting off for the day, so you are welcome to make the PR today if you want @salhus. Otherwise, I can do it on Monday.

closed 2021-09-15 by kmruehl