WEC-Sim/WEC-Sim #695: [Theory or Implementation]

hossein-mehdipour, 2021-09-01:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
I am trying to do a sensitivity analysis for the values of damping and stiffness of PTO for the OSWEC model, but I'm not sure if my results are logical (and not too large).
I couldn't find any benchmark studies for the WEC-Sim OSWEC model, the next best thing I could find was [this study](https://asmedigitalcollection.asme.org/OMAE/proceedings-abstract/OMAE2014/V09BT09A048/278995), which has the model with the same dimensions but I think it's a floating WEC so its results for power output are different from mine.
There is a screenshot of the results of my 6 runs that I compared with this study which I attached to this message.
I'm using the OSWEC tutorial files, and only changed the wave type (and of course PTO K and C )
I wanted to ask you if there is any benchmark cases I could run to make sure of my results, if not I was wondering if you could check my table and tell me if you see anything wrong with the results.

**Describe the theory or implementation approach that you have a question**
N/A

**Personal project context**
N/A

**Describe the type of conclusion or resolution you are looking for**
N/A

**Additional context**
[image]

nathanmtom, 2021-09-07:
@hossein-mehdipour Thank you for your question and I will do my best to try and answer. 

The Yu 2014 paper has fixed and floating OSWEC results so if your dimensions are the same as Table 3 in the reference you should be able to use Figure 4 to compare published results against your own case. However, I do not believe in Yu 2014 that the restoring stiffness at the hinge is the PTO stiffness as just below equation 6 in the paper the lead author mentions that K_{PTO} was set to 0 so your third column should be 0.

The only other reference where the authors attempted to benchmark a bottom fixed OSWEC can be found [here](https://www.sciencedirect.com/science/article/pii/S0960148111005672?via%3Dihub), which is titled as "Numerical benchmarking study of a selection of wave energy converter."

Another way to check your answer would be to compare your mean power output to the power contained within the wave. This can be found in the `waves.Pw` variable which provides the power in Watts per meter wave crest. Therefore, dividing your mean power output by `waves.Pw` would give you a unit of length and we expect this value to be lower than the length of the device perpendicular to the wave crest.

I hope the responses above will help with continuing your work, but let me know if you still have questions or need greater clarification. 

Cheers, 
Nathan
National Renewable Energy Laboratory

hossein-mehdipour, 2021-09-20:
Hello @nathanmtom ,
I followed your last suggestion method, and calculated the ratio of the mean output power to waves.Pw. I'm using the default OSWEC model in wecSim which has a length of 18 m. I calculated this value for different cases and in one of them it reached 19.5 which is greater than 18. I wanted to ask you if this is possible or am I doing something wrong?

Note: I'm calculating the mean output power from the end of the ramptime to the end of the simulation time (in the default case of wecSim it's from 100 s to 400 s). When I use the full simulation time to calculate the mean output power I get the value of 15.1 for the ratio. I thought maybe this could be important.

nathanmtom, 2021-09-21:
@hossein-mehdipour Thank you for the follow-up on the question you opened this issue on and I'll do my best to respond to your latest set of questions. 

I'll need more information about the PTO parameters and the sea state details for the wave environment that you simulated to obtain a capture width of 19.5 m. But in theory it is possible for WECs to have a capture width larger than its characteristic length and suggest reviewing this paper **[HERE](https://doi.org/10.1016/j.renene.2015.02.049)**. This has to do with the radiation pattern of a WEC being able to interact with the waves to either side. However, this theoretical upper limit generally requires follow the inviscid assumption, have no cap on the oscillatory motion of the device, and larger motions do not create other nonlinear forcing. So in the end, yes it is possible but often it is difficult to achieve in practice. 

To your second point, the ramp up period scales the wave input up from zero and allows the model to start from zero initial conditions and reach an approximate steady state. Therefore, it is more accurate to calculate the mean output power from the end of the ramp time to the end of the simulation. 

I hope this helps answer your questions, but please let us know if questions still remain. 

Cheers, 
Nathan
National Renewable Energy Laboratory

h7ossein, 2023-03-01:
Hello again @nathanmtom , so it took me two years to set the PTO stiffness to zero, but I have finally done it. But the results aren't much better. I wonder if you have any additional thoughts on the difference in power output.
[image]

nathanmtom, 2023-03-01:
@h7ossein The only additional piece of information I could provide is that if you are using a jonswap spectrum, how many different sea state realizations did you use? Ideally if you ran the model long enough, likely more than 400 s, the mean power output would converge towards the same values. If you and the Yu2014 paper are using different seeds or different discretization approaches you could see some of the variations you are observing. Unfortunately without having the original files from Yu2014 it is going to be difficult to track down other modeling mismatches that you are showing in the table. 

I assume that you would want the values to be closer, but from a reviewer perspective the values show the same trend in power with sea states which is encouraging.