WEC-Sim/WEC-Sim #699: [WEC-Sim Applications] RM3 HPTO power output weird behavior

hossein-mehdipour, 2021-09-10:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
PTO-Sim

**Describe the bug or issue**
I have 2 cases of an HPTO with almost equal parameters, but this small change in the inputs makes a huge change in the mean power output (183 & 281 kW). I was wondering why this happens?
These cases have almost the same value for absorbed power.
I ploted the power plots, and I noticed that in the Case A, after almost 300s into the simulation, the generated power suddenly drops, and that makes a huge difference in the power output. I would really appreciate it if you could help me understand why this happens.

**To Reproduce**

1. Use the RM3 non-compressible fluid HPTO folder, just change the parameters in the screenshot below.
2. Plot the absorberd power and generated power in the` userDefinedFunctions.m`.

**Expected behavior**
N/A

**Screenshots**
[image]
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 7]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v4.1]

**Additional context**
I'm using the RM3 case with non-compressible fluid, and using the default parameters for everything except these 4 inputs and wave H, T & Spectrum (it is shown in the table screenshot above)

jleonqu, 2021-09-15:
Hello @hossein-mehdipour 

Can you upload your PTO-Sim input file?

hossein-mehdipour, 2021-09-18:
Hi @jleonqu , 
I think I found out what my problem was. In case A around 300s mark, the delta-pressure (`u`) in the hydraulic motor exceeded the pre-defined value in the `variableMotorVolume.m` so the value of `alphaD` became very small just as the code intended.

Does this make sense? If not I will upload the `ptoSimInputFile.m` for you to take another look.

Thank you

jleonqu, 2021-09-20:
@hossein-mehdipour 
Make sense. That was my initial thought. I wanted to see your files just to confirm. I am glad you found the issue.

closed 2021-09-20 by hossein-mehdipour