WEC-Sim/WEC-Sim #700: [Theory or Implementation] Different results when using WEC-Sim directly Vs. in a function

hossein-mehdipour, 2021-09-13:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
I am using the OSWEC example with the Linear PTO (the default WEC-Sim). When I run `wecSim `directly (by changing the values in `wecSimInputFile.m`) vs. when I run it in a function, I get different values for power and other outputs which I find very confusing. I remember doing a similar project before, and I didn't have such a problem. I was wondering if you could help me solve my problem.

**Describe the theory or implementation approach that you have a question**
N/A

**Personal project context**
I was just trying to check if my results are logical and this issue happened, and made me think maybe there's an underlying problem with the code that I'm missing.

**Describe the type of conclusion or resolution you are looking for**
N/A

**Additional context**
N/A

akeeste, 2021-09-13:
Hi @hossein-mehdipour 

Thanks for following up and using our issue template. 

I recall that your similar issue in #646 was resolved by PR #632. Can you provide more information on how you are using WEC-Sim in a function? Both ``wecSim`` and ``wecSimFcn`` should give the same results. However maybe concurrent runs with ``wecSimFcn`` results in the output scripts seeing the wrong data. Are you using the provided ``wecSimFcn.m`` file or your own custom version? Please upload your case files if possible.

Thanks,
Adam

hossein-mehdipour, 2021-09-14:
Hello again @akeeste 

I'm not using `wecSimFcn`, but I'm using a very simple code I wrote which you can see below. (I've simplified the code for the purpose of this issue)

[image]

needless to say I assigned the four inputs of this function (`H `, `T `, `K `& `C `) in the `wecSimInputFile.m`.

Let me know if you need any more details you want me to share.

akeeste, 2021-09-14:
@hossein-mehdipour 

> I assigned the four inputs of this function (`H `, `T `, `K `& `C `) in the `wecSimInputFile.m`.

If you define these inputs in the wecSimInputFile, you may be getting some circular behavior. For example, if you change any input value between lines 1 and 3 of the above function, they will be overwritten. WEC-Sim itself will run the input file and overwrite H,T, K, C at line 3 of this function. I suggest printing the values of H, T, K, C immediately and after your WEC-Sim call to see if they are being changed from what you expect. 

If you do not modify the input values before calling ``wecSim``, I suggest that you remove them as inputs entirely. Then the only place where they are defined is when ``wecSim`` runs the input file. In this way you will avoid any kind of overwriting.

Let me know if this helps solve your problem.

Adam

akeeste, 2021-09-21:
Hi @hossein-mehdipour 

I wanted to check in and see how this issue is progressing. Were you able to resolve your problem yet?

Thanks
Adam

hossein-mehdipour, 2021-09-22:
Hello @akeeste 

Thank you for following up. I think my problem is gone now, I'm not completely sure and I don't really know how it got resolved, but I think it was because minute difference in my codes. Anyway I will close this issue for now, but if I found out more about this problem, I will comment on this issue.

Thank you again
Hossein

closed 2021-09-22 by hossein-mehdipour