WEC-Sim/WEC-Sim #708: [Theory or Implementation] Power output

haniyehtp, 2021-09-19:
**Is your question request related to a problem? Please describe.**
I’m trying to model a FOSWEC. The characteristics are based on “The NumWEC project” paper. But the produced power is 20% of the expected power based on the paper. I’m not sure what is the exact problem with my model and I hope your team can help me with this problem. 

**Describe the theory or implementation approach that you have a question**
My model has 4 flaps and a floating frame. For the BEM data, I used Ansys Aqwa and I assumed that the mass of bodies is equal to their displacement. I modeled the 5 bodies separately in Ansys Aqwa and then I combined them with BEMIO. I ran the WEC-SIM code with 4 mooring lines and I just assumed a 40 kn/m stiffness with the mooring matrix. The wave states and the PTO damping and stiffness were based on “The NumWEC project” paper. But the power is much less than expected. For the irregular wave condition of JONSWAP  and the wave state of 2.5 m wave height and 11 s wave period, the average power is 16 kW in WEC-SIM whereas the expected power is 80 kW in “The NumWEC project” paper.

But the thing is when I assume that the frame is fixed the produced power increase a lot and it's about 50% more than the expected power in “The NumWEC project” paper. I guess that maybe it has to do with the mooring properties. I thought about modeling the mooring with MoorDyn but I figured that I can’t model more than 1 mooring line with MoorDyn. Is that right?

And I have a question about the WaveSpread setting. For one single wave direction angle of 0 degrees, what the value of WaveSpread should be? The maximum value for WaveSpread is 1? when I assign a higher value of  WaveSpread the power gets higher. But I don’t know if WaveSpread value can be more than 1 or not.

And I wanted to know about the moment of inertia and which point it is calculated about? At the COG of the body? Or at the hinge?

**Personal project context**
Here I share my WEC-SIM files.

[FOSWEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/7191694/FOSWEC.zip)

**Describe the type of conclusion or resolution you are looking for**
I don’t understand why the power produced in my model is much less than the expected power of 80 KW. Thank you in advance for your support.

**Additional context**
N/A

Regards,
Haniyeh

dforbush2, 2021-09-24:
Hi @haniyehtp 

Can you please provide a *zip file of your input file, hydrodata, and simulink model please? The screenshots are helpful but the ability to run the model from my end would be ideal.

There are a bunch of questions here so I will try to break them out one by one:
1). "Modeled each body separately in ANSYS and combined them using BEMIO"
I am not familiar with the NumWEC paper, but have worked extensively with the FOSWEC, and I am dubious of this implementation because a strong hydrodynamic coupling exists between the flaps. I am not sure that running ANSYS on the separate bodies will capture this coupling. Additionally, assuming that the mass of each body is equal to displacement is potentially problematic, since the aggregate 5-body system floats at the prescribed position, but each individual body is not neutrally buoyant. This alone is probably introducing a significant difference between your results and the NumWEC paper. 
2). Your mooring matrix implementation is atypical: the mooring matrix simply assumes a linear restoring force that is applied proportional to displacement and velocity of the connected body (F_moor(i) = mooring(i).matrix.k * displacement + mooring(i).matrix * velocity ,  in the 6 DOFs). The "other end" of the mooring is assumed to be some stationary point such that the displacement of the mooring is equal to the displacement of the connected body.  Since each line is connected to the same rigid body (I assume?) you should be able to condense your 4 mooring lines to a single, equivalent connection. 
3). Can you model more than one line with MoorDyn? In short, yes. There are examples provided in their documentation with how to do this. 
4). "And I wanted to know about the moment of inertia and which point it is calculated about? At the COG of the body? Or at the hinge?" **All hydrodynamic parameters used by WEC-Sim MUST be calculated about the COG** This includes the moment of inertia. Simscape applies the necessary transformations based upon the constraint/PTO locations. 
5). The waveSpread setting. Check out the documentation for waveClass. http://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#user-advanced-features-wave. Wave spread is a probability vector of wave direction. So if the value is 1 and length 1, 100% of waves will come from the single prescribed direction. It should not be more than 1. Frankly we should probably incorporate an error message if this is specified to be larger than 1. 
6). "Can you give guidance on how to calculate the optimized PTO damping and stiffness for a wave state?" Oh boy can I. See closed issue #422 .

haniyehtp, 2021-09-27:
@dforbush2 Thank you for your help.
Here are my model’s files.
[model.zip](https://github.com/WEC-Sim/WEC-Sim/files/7235488/model.zip)

1) The reason I modeled the bodies separately was that the Ansys Aqwa has a limitation for the number of elements for meshing, and due to the large size of my model I couldn't model the whole bodies in one model. You said there is a strong hydrodynamic coupling between the flaps, so do you think that I can model the 4 flaps in one model and the frame in another model? Can it give a more realistic result? Cause I can’t model the whole frame and flaps in one model.
About the mass of the bodies, I considered it equal to the displacement of the bodies because it was mentioned in the paper that I was using. 
 
But I did model the flaps with half of this mass and the frame with some ballasts but it did not work and it made the FOSWEC  unstable and it became floated on the water. I did model the bodies with the mass equal to displacement for them to be stable. Is there anything about the mass that should be considered here?
And here’s another thing about my model. I assumed the water size in my BEM model 100meter x 100meter. Now I understand that maybe it is small compared to my model size of 25x25 and it can make some problems in boundary conditions. Is it so? And is there any formula for considering the water size?
2) about the mooring matrix I wanted to define stiffness in translational DOF, so I looked in the issue [https://github.com/WEC-Sim/WEC-Sim/issues/43](url) and concluded this kind of matrix implementation. If I want the mooring to have stiffness in horizontal and vertical directions how can I define it?
3) thanks, I will look it up in the user manual.
5) yeah, so Wave spread can’t be more than 1.
6) thank you so much, I’ll read the papers for PTO coefficients.
It will be a great help if you check out my file and help me with the mistakes in my modeling.
Regards,
Haniyeh

haniyehtp, 2021-10-18:
Hi @dforbush2  
Hope you're having a great day.
I will be so grateful if you help me with my questions. Especially about the mooring matrix and mass of the bodies and their stability. 
Thank you in advance
Haniyeh

yuyihsiang, 2021-11-04:
Hi @haniyehtp 

First of all, please take a look at our [METS2014 paper
](http://vtechworks.lib.vt.edu/bitstream/handle/10919/49236/137-Lawson.pdf?sequence=1&isAllowed=y
). We had an example of using WEC-SIm to model the Langlee design presented by Babarit in the NumWEC project and compared our solutions to his results.

To answer your question:

1. The purpose of the mooring here for the Langlee design is to keep the device in position. The mooring should have VERY little influence on the WEC power performance in this case. In our METS2014 paper, we used ONE mooring matrix with a small spring stiffness value of 40 kN/m. There is no need of having four mooring matrix. 
2. If you lock the frame, it is expected to generate more power because the system will loose power when the frame moves, except for some special resonant wave frequencies.  
3. For WEC-Sim, the moment of inertia is always given at the COB. 
4. One thing I noticed is that you had 4 flaps and 4 PTOs in your model, but we had 2 flaps and 2 PTOs in our METS2014 study. I believe this is what Babarit did as well. In our case, we modeled the two fore flaps as one body and two aft flaps as another body. This assumption is valid when the wave direction is normal to the flap surface. If you model the system using four flaps, you will need to cut your CPTO and KPTO values by half. 
5. As @dforbush2 have said, modeling the bodies separately and combine them in BEMIO may result in  a set of inaccurate hydrodynamic coefficients, where the interaction between bodies are neglected.   

--Yi-Hsiang

closed 2021-11-17 by kmruehl