WEC-Sim/WEC-Sim #709: [Developer Issue]

akeeste, 2021-09-20:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
WEC-Sim Roadmap

**Describe the issue**
Supplied userDefinedFunctions scripts have not been updated for the new output.plotResponse() function (a body input is required for translational directions).

Files that need updating:
examples/OSWEC
examples/RM3FromSimulink
tutorials/RM3
tutorials/OSWEC
All WEC-Sim Applications repo examples that use plotResponse

kmruehl, 2021-09-21:
Good catch. I think we need to make a revision to https://github.com/WEC-Sim/WEC-Sim/pull/698 so we don't need to pass `body`. For example, we could consider adding `body.cg` to the WEC-Sim  `output` object.

akeeste, 2021-09-21:
@kmruehl agreed. This would keep it backwards compatible and easier to maintain.

commit a56c9649 (2021-09-22): Resolves #709 to fix plotResponse by passing cg to output  (#711)

* resolving #709 by passing cg to output

* adding cg to bodiesOutput

* comment plotWaves in OSWEC example
closed 2021-09-22 by kmruehl