WEC-Sim/WEC-Sim #715: [Developer Issue] Which branch should WEC-Sim_Applications test against?

H0R5E, 2021-09-23:
## Project

WEC-Sim_Applications

## Describe the issue

I am running into some difficulties preparing the CI for WEC-Sim_Applications because changes are made to that repository that are only reflected in the `dev` branch of WEC-Sim, not in the `master` branch. See the test failures [here](https://github.com/H0R5E/WEC-Sim_Applications/runs/3688750196) due to the introduction of the `nlHydro` property in the `bodyClass`, for instance.

So my question is should WEC-Sim_Applications only work for the WEC-Sim `dev` branch? If it supposed to work for both `dev` and `master`, then perhaps we should add a `dev` branch to the WEC-Sim_Applications repo as well?

akeeste, 2021-09-23:
@H0R5E I like the idea of maintaining both ``dev`` and ``master`` branches of the Applications repo. We have also discussed the idea of releasing versions of the Applications repo that correspond to each WEC-Sim version. Our upcoming WEC-Sim release next week would be a good opportunity to make this switch.

H0R5E, 2021-09-23:
@akeeste, yeah, I think it's the most sensible approach, given that changes are being made in-situ with changes on the `dev` branch here.