WEC-Sim/WEC-Sim #721: Regular rot PTO motion + PTO motion actuator

t-osu, 2021-09-30:
Morning all,

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Use of PTO motion actuator_

**Describe the bug or issue**
_Attempting to add control with delayed motion feedback. Requires addition to the usual rotational PTO motion, but when use motion actuation it only passes through the position/velocity/acceleration inputted, which is zero because isn't defined_

**To Reproduce**
_Steps to reproduce the behavior:_
There isn't really an example file, but you can go to the OSWEC example and change the rotational PTO to a rotational PTO actuation motion and not include any inputs. When you do this, the OSWEC does not move at all, because there is no input. When you do this for the rotational PTO actuation torque, it will move as if the normal rotational PTO were attached b/c it's just additional torque, however I am using previous values of position to influence future behavior, leading to use of the motion actuator, not the torque. 

Maybe a simpler way to explain the problem would be, the output one would get from a rotational PTO, but plus 2 radians however I do not want to just add this to the response side of the PTO (where it gets exported to the output variable) as this would not have influence on the overall system simulation. It needs to be somewhere in the base/follower signals. I've tried a few things with no success (2 PTOs, 2 joint revolutes).

**Expected behavior**
_Regular motion when using rotational PTO, with the addition of previous position values influencing future values _

**Screenshots**
Will include what I've tried if needed.

**Workstation Information (this section MUST BE completed):**
OS: Microsoft Windows 10 Enterprise Version 10.0 (Build 19042)
[MATLAB/Simulink Release] MATLAB Version: 9.9.0.1592791 (R2020b) Update 5...Simulink Version 10.2
[WEC-Sim Release] v4.2

**Additional context**
Thoughts? Thank you.

nathanmtom, 2021-10-04:
@t-osu Thank you for brining this issue to the WEC-Sim Development Team's attention. I'll do my best to try respond and answer your question(s).

After reading you issue above, let me ask a few clarifying questions to make sure we understand your situation. 

1. You want to use the rotational PTO with actuation motion rather than actuation torque. 
2. You require two signals, one is the current rotational angle from the rotational PTO and the second is that same signal except adding 2 rad or a static constant offset. 
3. You want to use the rotational position, velocity, and acceleration at time (k) to calculate the response at time (k+1). 

Clarification on observed issues:

1. You are unable to get the rotational PTO actuation motion block to work?
2. You are unable to get the expected rotation of the PTO in regular waves? 

It would be appreciated if you could add a few screenshots or graphics to help illustrate where in the work flow `Re: somewhere in the base/follower signals` that you believe or need the updates.

Cheers, 
Nathan
National Renewable Energy Laboratory

t-osu, 2021-10-05:
Morning,

Questions answered:
1. Yes
2. Yes
3. Yes. A little more in depth, I would like something such as t(k+1) = t(k) + t(k-2). So the current and the past to determine the future.

Clarifying issues:
1. It is working as it normally should, you input a position signal and it will move according to that signal. When I try to use a delay block though, the position starts from 0, and therefore will continue to stay at 0. The motion actuation doesn't really allow "passive" motion unlike when you use torque. If you input no torque, it will move like a PTO with no damping. I feel I can't get that with the motion actuation since it needs the exact position/vel/acc signal to move. An idea I was thinking of was running a simulation first without the actuation to obtain the position data, then importing that using a "from workspace" block for a simulation with the delays so it has a place to start from?
2. I guess it would more so be this, can't get it to work as expected.

Picture:
Attached below is inside the motion actuator block. I know that I cannot just add a constant block of 2 to the position signal 'q' after the joint revolute block as this has no affect on the simulation and will only move the output file of position values up by a value of 2. So I feel it has to be somewhere in the base/follower signals to actually have an influence on the simulation.
[image]

nathanmtom, 2021-10-05:
@t-osu Thank you for the clarifications and the additional screenshot. 

I apologize as it wasn't clear in my first response, but you want to use past position/velocity/acceleration measurements to calculate the future position/velocity/acceleration of you pendulum, correct? 

If so, so it possible for you to export the motion response, the signals you have circled in blue, similar for the force actuation block, see below in red outline. This would give you the output from the block which you can then store, manipulate, as you see fit before sending back as the input to the rotational motion actuation block. This seems to circumvent your issue with the delay block while getting past information to complete your control implementation.

If you want the natural motion of the WEC and pendulum then the rotational motion actuation block does not seem appropriate as you will have to define the inputs or else the connection will be rigid. Therefore, although running and then storing the time history of a previous run should work, the issue seems to be storing enough starting data, t(k=0), t(k=1), t(k=2) to initialize your control scheme. This could be done by initializing a variable in the workspace that stores this data and allows WEC-Sim to use.

I hope this response provides you additional feedback and ideas, but let me know if I have misunderstood any of your comments or this still doesn't answer your question. 

[image]

Cheers, 
Nathan
National Renewable Energy Laboratory

t-osu, 2021-10-05:
Yes, you are correct.

I actually did try this yesterday, exporting the motion as is done in the force/torque actuation blocks, but was running into the issue of the fact that is has no initial motion so all position values stay at 0.

I will try the idea of running a previous simulation as you agree with its functionality purpose.
Thank you!

closed 2021-10-05 by t-osu