WEC-Sim/WEC-Sim #726: [Wave Gauge to Measure the effects of a WEC on its surroundings]

timmaflanagan, 2021-10-05:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
I am trying to observe the effect that a WEC will have on its surroundings. It would be ideal if there was a way to use the wave gauge to recreate the wave field behind the WEC. I understand that you have a "wavegauge" feature, however I don't understand how this can be used. Will it tell you the free surface elevation in front of and behind a WEC? or does it just replicate the same time series as the origin?  

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
a time series of the wave behaviour in different locations relating to the device would be great but I understand that this might not be possible in WEC-Sim? 

**Describe alternatives you've considered**
 I have considered estimating the wave height behind the WEC (OSWEC/RM5) using the relationship of power capture to wave height. (P=0.5H^2*Te, or sqrt(P) is proportional to H, if 20% of the wave power is extracted, the wave height will decrease by 4%.

**Personal project context**
I am trying to couple XBeach and WEC-Sim. XBeach has the capability to measure how wave action will effect sediment change, as well as run up etc. currently i am using a soft coupling method, just based on equations known about power capture, however I would love to be able to obtain a wave surface elevation directly from WEC-Sim and put it into XBeach. 

**Additional context**

nathanmtom, 2021-10-05:
@timmaflanagan Thank you for your question and I will do my best to answer your question. 

Unfortunately, WEC-Sim does not have the capability to show how the presence of the WEC will have on the surrounding wave field. The wave gauge options are plotting the undisturbed incident wave field, but does not modify the wave field around the WEC to include the radiation and diffraction wave fields. The wave gauge option was developed for the Wave Energy Converter Control Competition (WECCCOMP) where competitors wanted to use the information from an up-wave measurement device to determine their control action. 

Therefore you can place wave gauges at different points in the X-Y plane, but they will still only be the incident wave potential without any energy removed from the wave system. The alternative you have considered could be a reasonable approximation if you are modifying the wave field several wave lengths behind the WEC where you could assume that all transient effects have died off and the remaining wave field is planar propagating waves; however, I believe if you want a 20% reduction in wave power this would correspond to a roughly 10% decrease in wave height as you would be taking the square root of 80%. 

I hope this answers your initial question, but please let us know if you need clarification or have other questions. 

Cheers, 
Nathan
National Renewable Energy Laboratory

timmaflanagan, 2021-10-05:
Thank you! You are correct on the math. Thanks for catching my mistake.

closed 2021-10-05 by timmaflanagan