WEC-Sim/WEC-Sim #762: Calculation of Net Power with PTO actuators

t-osu, 2021-11-03:
Afternoon,

**Describe the bug or issue**
Having trouble figuring out a way to calculate net power output if torque actuator is used, or is this already included through the revolute joint? If control is used, power is needed for it so I am trying to see where I could find the power associated with the control vs the power that is generated. I saw the total torque and torque constraint signals with the joint, but I'm unsure on their exact meaning.

**To Reproduce**

1. Add a rotational torque actuator to OSWEC example. 
2. Set a sine wave source to the torque input. Terminate the response output.
3. Log actuated torque and torque from internal mechanics.

**Expected behavior**
OSWEC runs with addition of actuated torque. 

**Screenshots**
N/A

**Workstation Information (this section MUST BE completed):**
OS: Windows 10
MATLAB Version: 9.11.0.1769968 (R2021b)
wecsim v4.4

**Additional context**
Would the net torque be the one calculated from internal mechanics minus the actuated torque? Then multiplied by speed for net power?

Thanks.

nathanmtom, 2021-11-05:
@t-osu Thank you for your question and I'll do my best to answer your question. 

If I have read your question correctly, you are using the `Rotational PTO Actuation Torque` block within WEC-Sim and you want to calculate the power either generated or consumed by your PTO. To either pull power out of the rotation or speed up rotation to maintain a given velocity or position. 

I am assuming that you are not using the `pto.c` or `pto.k` options within the `ptoClass` and you are feeding in a desired torque you calculate for the rotational PTO at each given time step. The `output.ptos.forceInternalMechanics` is calculated only when `pto.c` and `pto.k` are non-zero, which for now I've assumed you have set to 0 based on your issue description. 

If this is indeed the case, then the instantaneous power that is generated or consumed by the PTO would be the rotational velocity of the PTO times the PTO torque (`output.ptos.velocity` x `output.ptos.forceActuation`). If the signal is negative you are extracting power and if the signal is positive you are inputting power into the system.

I hope this helps answer your question, but please let me know if I have misunderstood your question or modeling set-up.

Cheers, 
Nathan
National Renewable Energy Laboratory

t-osu, 2021-11-05:
> [This email originated from outside of OSU. Use caution with links and
> attachments.]

Thanks Nathan.
What would happen if one were to have a pto.c input? Would the total force
be the signal to multiply by pto velocity for the net power?

nathanmtom, 2021-11-08:
@t-osu To clarify, you are not using the torque actuation block anymore and replacing with a `pto.c` linear rotational damping coefficient? If so then yes, the force from the PTO would be the rotational velocity of the pto times `pto.c` to obtain the resistive torque of the PTO. 

If you are having a `pto.c` and a PTO torque actuation block, I'd have to ask for your reason for having both? In the pto torque actuation block you are defining a desired torque of the PTO regardless of the PTO rotational velocity, while having the `pto.c` coefficient generates a resistive torque that is linearly proportional to the PTO rotational velocity. 

I hope this helps answer your question, but please let me know if you need more clarification or have other questions about this topic area. 

Cheers, 
Nathan

t-osu, 2021-11-12:
I think my confusion stems from the fact that I am not using the PTO torque
actuation block to achieve desired torques, rather I'm trying to use it to
input supplementary torques. I noticed if you have an input of 0 for the
torque actuation PTO block, it gives identical results to if only the
rotational PTO block were used, leading me to think that if you had an
input with a value, it could be seen as adding an additional torque to the
natural movement? Like if you had a PWM signal attached to it, then that
amplitude on the ON cycle would be similar to giving the PTO an additional
torque exerted on it or no?
If it doesn't, would there be a way to achieve this? I tried putting a
rotational PTO and torque actuated PTO in series and it did not work.
Or, if my thinking on that is right, and that is how I intend on using it,
I don't have a specific torque that I'm trying to achieve, leading me to
keep a pto.c value since it isn't interfering with any desired values.

According to documentation from matlab, the total torque "is a sum across
all joint primitives over all sources—*actuation inputs, internal springs
and dampers,* joint position limits, and kinematic constraints" which is
why I think I would need to use the total torque signal vs the actuation
signal when calculating power.
Thanks for the help,
- Chris

nathanmtom, 2021-11-15:
@t-osu Thank you for the additional context and let me make some clarifying remarks to see if I follow what you have done and trying to do.

1. You are using the PTO torque actuation block, but setting a `pto.c' so that when you do not set an input torque the response of PTO responds as a passive linear rotational damper. 
2. You plan to use the input torque actuation feature to maintain a certain set of rotation characteristics, but this will not be active at all time steps.  

If I have understood the description correctly, my question would be why not implement the `pto.c` term in the same subsystem as your actuated torque? The Rotational PTO actuation block already outputs the rotational response of the PTO and you can take that time signal, multiply it by your desired linear PTO damping coefficient and then send into the torque block input. In your subsystem you are then able to add any additional torque need on your system to achieve you desired response. 

This way you can use the PTO actuation torque signal and multiply against the PTO rotational velocity to obtain your instantaneous output. Or you can save the PTO torque contributions to your desktop to complete analysis assuming the two torque contributions are separate. I'll point you to the WECCCOMP example in our WEC-Sim Applications folder to see how this could be implemented. 

Cheers, 
Nathan
National Renewable Energy Laboratory

t-osu, 2021-11-15:
To see if I'm following right, would something like this be what you mean?

[image]

nathanmtom, 2021-11-16:
@t-osu Yes, your proportional damping torque and external torque are calculated externally to the PTO actuator block and you can calculate their individual proportions to the instantaneous power at any given time. However, does this suit your needs?

Cheers, 
Nathan
National Renewable Energy Laboratory

t-osu, 2021-11-17:
A couple of things I'm still unsure on.
One is still on why you can't have an actuated torque and a pto.c defined
within the joint. I understand if you had a specific torque you wanted to
apply to the PTO, you wouldn't want it to be affected by a damper on the
joint.
But I think my application is different. I am inputting some values (e.g.
constant, PWM, sine wave, signal generator, etc) and watching the response
the PTO has and if it is displaying certain characteristics we are seeking.
As you suggested earlier, if I were to have a pto.c term, why not put it on
the outside like I had done in the image above to calculate their
respective portions to the instantaneous power. This doesn't dampen the
joint though and the response will be different.
So is it wrong/incorrect to define a pto.c in the joint while having an
actuator torque? Or is it more so something based on how you are using it?
The use of the pto.c has a major impact on the response of the system,
including if the PTO moves at all. I had no pto.c defined and a constant
torque of 2, and it resulted in 0 torque in the forceinternalmechanics.
Whereas with a pto.c defined in the joint, the joint moved much more.

Another thing I'm wondering is what the difference between calculation of
torque is. If there is a constant PTO torque actuation and a pto.c in the
joint, say a value of 2 being inputted, then the torque total and joint
response signals will be affected. If you look at the torque total signal
compared to the forceinternalmechanics signal, they are off by an addition
of 2 (forceinternal+2 = torquetotal). Which is the true torque one should
be using to calculate power, then? Both are affected by actuation and
damping, but the velocity * pto.c versus however simscape calculates torque
total, yields different values.

Thanks for your continuing help Nathan

nathanmtom, 2021-11-17:
@t-osu Thank you for the additional feedback and comments.

In theory, the two should be equivalent, but the image above would give you the most freedom in controlling and separating the contributions from the linear rotational damper and the external input torque. 

Reviewing your image again two things 1) you have a broken library link which might imply you made other edits to the block that is affecting performance and 2) you will want to set your linear damping coefficient to -15. In the PTO block when using the `pto.c` the sign is automatically handled internally. So in your image you would be adding a torque in the direction of positive rotation. Perhaps try that as an option before determining your path forward. 

I could not follow your question in the last paragraph, but I believe this is getting back to my point earlier about tracking where your external actuated torque and the torque from the `pto.c` contributions are being stored in WEC-Sim's `output` class. If you have a `pto.c` in a rotational block, then part of the contribution comes from the force internal mechanics and force actuation, which should add to total force.

I hope this has helped clarify you latest questions or confusion. 

Cheers, 
Nathan
National Renewable Energy Laboratory

t-osu, 2021-11-18:
Thanks Nathan, I did some cases with the pto.c defined in the joint and defined using the gain on the outside and got similar results. I have a better understanding of how torques are handled and power is calculated.

closed 2021-11-18 by t-osu