WEC-Sim/WEC-Sim #763: [BUG] Error when generating .h5 for a two body system

vibishana, 2021-11-08:
**Describe the bug**
Hello! I'm trying to model a bottom fixed point absorber(cylydrical buoy) and succeeded in doing so by placing a translational PTO in between two rotational joints; one near the sea floor and one at the water surface. Now I want to place a second body(sphere) between the PTO and topmost rotational joint. I did the hydrodynamic analysis using NEMOH for two bodies separately, manually edited Nemoh.cal and renamed the files _0 and _1 accordingly. However, when I'm trying to write .h5 file bemio returns the following error.

**WEC-Sim file(s)*

[TwoBodySys.zip](https://github.com/WEC-Sim/WEC-Sim/files/7495800/TwoBodySys.zip)

**To Reproduce**
run bemio

**Expected behavior**
To generate .h5 file.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
OS: [Windows 10]
[MATLAB/Simulink Release][2020a]
[WEC-Sim Release][v4.3]

**Additional context**
N/A

nathanmtom, 2021-11-09:
@vibishana Thank you for your question and I'll do my best to try and answer your question. 

First off, may I ask why you ran two NEMOH runs with separate bodies? If the bodies are close to one another than you will not be accounting for any body-to-body interactions. Therefore, we generally encourage you to run NEMOH or your hydrodynamic solver with both bodies included. 

If you must use two different NEMOH runs then I'd suggest using our [Combine_BEM.m](https://github.com/WEC-Sim/WEC-Sim/blob/master/source/functions/BEMIO/Combine_BEM.m) or consider using the [create_h5File.m](https://github.com/WEC-Sim/WEC-Sim_Applications/blob/master/write_hdf5/create_h5File.m) which should be easier than editing the NEMOH while ensuring the proper formatting that BEMIO is expecting. 

Yes, by setting the mass property to equilibrium will set the mass of the body equal to the displaced volume of the body times the fluid density. If your body is completely submerged then the mass and buoyancy forces should cancel out. 

I hope my answers help you to continue developing your model, but let me know if you need clarification or have additional questions.

Cheers, 
Nathan
National Renewable Energy Laboratory

vibishana, 2021-11-09:
@nathanmtom Because I didn't know that you can run NEMOH for two bodies at once. I will try to run with both bodies included and let you know if I encountered any problems. Thank you!!

nathanmtom, 2021-11-09:
@vibishana Thank you for your quick response! I will only add that we have examples of two body NEMOH runs in our example folder. I'll direct you the following RM3 example: 

https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO/NEMOH/RM3

Cheers, 
Nathan
National Renewable Energy Laboratory

vibishana, 2021-11-09:
@nathanmtom Thank you! I will check it out.

vibishana, 2021-11-09:
@nathanmtom  Could you also let me know the most convenient way to convert a mesh generated in CAD software (say from Rhino) to NEMOH format? I read about Python based bemio but I'm not familiar with Python. Thank you!

nathanmtom, 2021-11-09:
@vibishana Unfortunately I am not a meshing expert and hope that @dav-og or @salhus might be able to provide some additional guidance. However, I would point you to [meshmagick](https://github.com/LHEEA/meshmagick) which is a tool the WEC-Sim team uses to convert and edit meshes to be used in various BEM solvers. 

Cheers, 
Nathan
National Renewable Energy Laboratory

salhus, 2021-11-09:
Hi @vibishana =,
I think meshmagick is a very good tool but requires commands voa little command line interface.
Sounds like you would prefer a more GUI.
I would suggest using Rosetta: https://github.com/BEMRosetta/BEMRosetta
The installation is quite straightforward.
1.  You just need to clone the repo,
2. No compiling is required. Just copy bin/BEMRosetta.exe and bin/BEMRosetta_cl.exe anywhere in your computer and dodge Windows warnings.

This tool would let you change mesh file formats, manipulate meshes, and also plot BEM results. Let me know if you need any more details or have any questions.

vibishana, 2021-11-10:
Thank you!! @nathanmtom and @salhus for your support. I started using Rosetta and let you know if I encountered any issues.

vibishana, 2021-11-10:
@salhus Thank you again for suggesting BEMRosetta. Let me give you a brief overview. 

My model consists of a floating cylindrical buoy tethered to a submerged spherical body of neutral buoyancy which then tethered to the seafloor through a translational PTO. I was able to generate a mesh in Rhino and import it into Rosetta.

When importing multiple bodies do I have to import them separately? and I'm not sure how to define COG each body separately or do I have to? 

Also are there any examples or tutorials available on Rosetta?

These are my meshes 
[RHINOmeshes.zip](https://github.com/WEC-Sim/WEC-Sim/files/7515772/RHINOmeshes.zip)

salhus, 2021-11-10:
Hi @vibishana ,

re:tuorials
I think the reame.md on their github page goes through some steps. It typically requires you to just identify the relevant file and then should be able to display results. Note, to state the obvious, each tab can read only the file format the Open window lets you.

re:multiple bodies,

If the assembly is always moving as a bulk you may could do the analysis as a single body. However, from what I gather this is not the case here. So have them as separate bodies, since they can have relative motion wrt each other. The centers of gravity are not defined in the gdf file (if using wamit). the gdf file does however defines the value of gravitational accn. g, the characteristic length, the number of panels and the coordinates of the panel vertices.

Similarly, Nemoh mesh files typically only define each panel's vertices,

https://lheea.ec-nantes.fr/valorisation/logiciels-et-brevets/nemoh-mesh#:~:text=Input%20files%20are%3A%20%E2%80%A2%20ID.dat%20file%3A%20identification%20file%2C,the%20body.%20It%20is%20normally%20made%20by%20hand.

re: Center of Gravity
The CoG is defined separate from the mesh files, (but is amongst the inputs for the BEM software packages). The CoG can be determined from the geometry when the material is homogenous. So if the buoy is made up of different materials, taking the volumteric center of the geometry may or may not be same as the overall CoG. So that's the reason the user is allowed to define the CoG independent of the mesh file for BEM runs.

re: defining the submersion behavior
The degree of submersion though related to the CoG is not determined by geometry or CoG. In fact, the BEM software never asks for the body density or mass. All hydrodynamic calculations for a BEM run are dependent on the volume of the water displaced and the pressure exerted at different discretized panels on the body. Although the CoG (and MoI depending on the BEM software used) )determines how the mass distribution displaces water, the hydrodynamics are still calculated based on the pressure exerted on the body due to fluid displacement.

When running WEC-Sim, or any other implementation of the time domain dynamics using Cummin's eqn, the user would have to define the body density and mass properties. So, here the user could define the body density \rho as greater than water (the body sinks), equal to water ( neutral buoyant), and less than water (body floats). Recall that the buoyancy force is the upwards force calculated based on displaced volume of the fluid. The body weight force acts downwards, so altering the body \rho would determine the extent of immersion. 
So for instance in your case when setting up wecsim, you would probably want to have the the second body that is neutrally buoyant to have the same mass as its volume \times the density of water, while the floating body should have its mass less than the displaced fluid.

Let me know if that makes sense

vibishana, 2021-11-12:
Hello! @salhus. Yes, I understood. Thank you for the explanation! I have a couple of questions regarding Rosetta.

1) As I understood BEM Solver in rosetta can be used to edit preexisting BEM outputs but cannot be used used to generate new ones using a mesh. Am I right about this? Because I tried loading meshes into BEM Solver but it did not produce all the files that NEMOH usually does in MATLAB. 

This is what it produced
[ROSETA Results.zip](https://github.com/WEC-Sim/WEC-Sim/files/7525446/ROSETA.Results.zip)

2) Then I converted my mesh into .gdf using rosetta and tried to use GDFMesh.m but it also returned the following error 
[image]

3) If I am to follow the second method (GDFMesh.m) how can I include both the bodies in the analysis?

Thank you in Advance!

salhus, 2021-11-13:
Hi @vibishana,

re: 1
Yes, you are right. It is not a GUI for runs a BEM run, rather a data explorer/navigator.

re: 2

Depends the BEM software you are using.
Looks like you are using NEMOH, so I would convert the .stl file directly to .dat file the NEMOH can read.
The Rosetta generated gdf file would work in Wamit but the way the nemoh's  gdfmesh function is written, it only works with gdf files that have all the 4 panel vertices in the same line. 

While Rosetta and rhino generate the .gdf file such that each vertex is on a new line. WAMIT  works either way.
So long story short, use Rosetta to convert to format that your BEM code needs.

vibishana, 2021-11-13:
Hello! @salhus 

What is confusing to me is how do I use that .dat output of Rosetta in NEMOH? Do I use it as an Input to Mesh.m? Can you please give me an outline of the workflow? 

Thank you in advance!

salhus, 2021-11-15:
hi @vibishana ,
I am not sure i understood that question.
The .dat file is like any other mesh input file that is used for NEMOH.
Workflow,
1. Generate BEM hydrodynamic coefficients ( refer the corresponding BEM manual). Note, if using multiple bodies that hydrodynamically interact with each other, the BEM run should have the bodies in the same run,
2. Run BEMIO to post-process the hydrodynamic coefficients, so that WEC-Sim can read it,
3. Setup WEC-Sim input file. Refer the Applications repo for some examples. 
4. Run WECSim
5. Post-process the WECSim results. The results are stored in the 'output' data structure. The user-defined functions file is a template that you can use to write you post-processing script.

I think the BEM related questions are better addressed by their community forums. 
However, if you find bugs in the WEC-Sim. please let us know.

Sal

vibishana, 2021-11-20:
Hello! I was able to run the WEC-sim model and will let you know if I encountered any issues. 
Thank You.

nathanmtom, 2021-12-03:
Hello @vibishana, there has not been any activity on this issue for two weeks now so we are going to close this issue. Please feel free to open up a new issue with any questions you have. 

Cheers,
Nathan
National Renewable Energy Laboratory

closed 2021-12-03 by nathanmtom

vibishana, 2021-12-05:
Hello! Previously I was working with MATLAB R2020a and due to compatibility issues (I suppose), the cable block wasn't visible to me.  So I used a translational PTO to represent the cable between two bodies (Cylinder and Sphere). Now I have upgraded to MATLAB R2021b and replaced the PTO with a cable block with translational constraints on either side. Unfortunately, it produces the following error constantly.  

1) When fixed time steps are used I get the following error. Run wecSim as it is to reproduce the error. 
[image]

2) When variable time steps are used simulation works although Cylinder moves independently and the cable does not have any effect on the submerged sphere. Maybe I'm not using the cable block correctly?

These are my files.
[POINT_ABSORBER.zip](https://github.com/WEC-Sim/WEC-Sim/files/7655419/POINT_ABSORBER.zip)

Thank you in advance.