WEC-Sim/WEC-Sim #766: How to extract the wanted hydrodynamics of a specified degree of freedom from WECCCOMP GitHub repository?

Doudou-Li, 2021-11-11:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
I think I want to build an optimal velocity [reference](https://ieeexplore.ieee.org/abstract/document/7737655) through the impedance matching idea for a control purpose. 
At this moment, maybe the pitch or heave motion radiation damping coefficient is the mainly term for my requirement to know. Therefore, I guess for this Wavestar numerical model, it is necessary to get its lumped damping term. But I am confused with how to obtain damping coefficients.
The WEC-SIM software provides 6-dof motions for a WEC system. I checked the .h5 file but it made me perplex.  I don't know how could I extract one degree of freedom data from that hydrodynamics file. 
There have many 6 by 6 matrices and I don't think it is right to just grab the [3, 3] index for heave motion or [5, 5] index for pitch motion. In addition, the data is very small in those 6 by 6 matrices, which I think this is due to the previous data pre processing normalized procedure.
Is there possible to build a function between WEC damping coefficient and frequency? Something like this [reference:](https://www.researchgate.net/publication/326607966_Study_of_Scale_Modelling_Verification_and_Control_of_a_Heaving_Point_Absorber_Wave_Energy_Converter)
[image]
In conclusion, I don't know how to get the proper information from a specified degree of freedom which I am interested.

**Describe the theory or implementation approach that you have a question**
In my view, for this scaled Wavestar modelling, the pitching is an artificial motion from the link arm are used for pto system.
I searched some references and found there have two or three papers employed a pitch moment equation as below picture shown:
[image]
I also tried to use the pitch motion to do wave excitation moment estimation which I think it is OK.
The theoretical value of excitation moment is calculated by 
[image]
from the [reference](https://www.sciencegate.app/document/10.1115/omae2019-95197)
However, could I consider to use heave motion to do the excitation force estimation and control works?
It seems that I couldn't find any other papers that shows the heave motion equation for the numerical model of WECCCOMP GitHub repository. Certainly, this also goes back to my problem on how to obtain useful hydrodynamics of one degree of freedom from WEC-SIM.

**Personal project context**
I haven't involved in any personal project, but just doing my PhD research study.

**Describe the type of conclusion or resolution you are looking for**
I'd like to consider resonance theory for power maximization design by using WECCCOMP GitHub repository, which provided a WEC-Sim model of the 1:20 scale Wavestar device. So it's necessary to build an [optimal velocity reference](https://ieeexplore.ieee.org/abstract/document/7737655) 
which is also based on one degree of freedom motion. Such as for a buoy, heave motion is significant.

**Additional context**
N/A

BR
Doudou

dforbush2, 2021-11-11:
Hi @Doudou-Li  

Some great questions here and I hope I can point you to some additional references. The fundamental issue you are citing is that the degree-of-freedom (DOF) of interest described by the actuator is not actually aligned with any of the DOFs described by the hydrodynamics? Is that correct?

In these cases, it is usually best to do some sort of system identification to figure out what the effective spring, mass, and damping are from the perspective of the PTO. I would recommend that you do this for both PTO-driven actuations (i.e., no waves, move the float via the actuator), and wave excitation. I would recommend following the clear procedure laid out in https://www.mdpi.com/1996-1073/10/4/472 that you should be able to replicate exactly. You might need to modify the WEC-Sim model to allow for the sort of open-loop excitation you will need to this procedure, but this should be straight forward. Your velocity of interest will be that of the PTO, which should be a direct output of the model, and the excitation force of interest is that exerted on the PTO.  

Personally, I find direct velocity feedback, like that discussed https://www.sciencedirect.com/science/article/pii/S2405896320315780 (see particularly the references of this paper) to be a simpler and effectively equivalent implementation than optimal velocity tracking, but I think your initial workflow looks the same either way.

Doudou-Li, 2021-11-13:
Hello,

Thank you so much for you very constructive suggestions. I will try to follow and read the references which you suggested here,
and to see whether I could treat wavestar prototype model as a heave buoy modelling in order to achieve resonance theory.

BR
Doudou

dforbush2, 2021-11-16:
Good luck. I am going to close this issue, but feel free to post another if more questions arise.

closed 2021-11-16 by dforbush2