WEC-Sim/WEC-Sim #767: How to show water surfaces and waters in mechanics Explorer

YU-XiaoYou, 2021-11-14:
I used my own model in wec-Sim, it is only one body. After inputing the command " wecSim", it seems good, but i can not see the water surfaces and water in mechanics Explorer. I want to see them. What do I need to do?

akeeste, 2021-11-15:
Hello @YU-XiaoYou 

Thank you for your question. We require users to open GitHub issues using an appropriate template. This is very useful to the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "Bug Report" template and include more details on what your errors, I am happy to help. Please maintain the template structure while completing the various sections.

Thanks!
Adam

akeeste, 2021-12-03:
Hi @YU-XiaoYou 

There has been no updates in two weeks so I will close this issue. Please open a new issue using a template if you have additional issues and I am happy to assist.

Thanks,
Adam

closed 2021-12-03 by akeeste