WEC-Sim/WEC-Sim #770: [BUG] Moving the library source files has changed install procedure

dforbush2, 2021-11-18:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
Splitting the library source files has changed the startup procedure. Following the procedure in the documentation results in several broken links

**WEC-Sim file(s)**
All library source files combined with startup.m. It may in particular be cableClass or cable blocks.

**To Reproduce**
Fresh clone the library and follow the install instructions as presented in documentation. Run a model with a cable block.

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
Command line from wecSimInputFile

**Expected behavior**
The body link in the cable class should initialize correctly from the startup procedure.

akeeste, 2021-11-24:
@dforbush2 See a fix in PR #774

dforbush2, 2021-11-24:
Confirmed that fix in PR #774 resolves this issue.

closed 2021-11-29 by akeeste