WEC-Sim/WEC-Sim #771: Developing plan

zash07, 2021-11-19:
Hello, 

Following a note that is mentioned in the advanced features manual “ WEC-Sim’s nonlinear buoyancy and Froude-Krylov wave excitation option may be used for regular or irregular waves but not with user-defined irregular waves.” I was wondering if you have any plan in near future to include the nonlinear buoyancy and Froude-Krylov wave excitation option even when using the user-defined irregular waves? 

Many thanks 
Best regards

akeeste, 2021-11-24:
Hi @zash07 

Thank you for your question. We require users to open GitHub issues using an appropriate template. This is useful to the development team to better classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please edit this issue to use the "Theory or Implementation" template and link to the relevant documentation that is unclear, I am happy to help.

Thanks!
Adam

akeeste, 2021-12-03:
Hi @zash07

Is there any more clarification you can provide on this issue? If not I will close this issue after another week.

Thanks,
Adam

kmruehl, 2021-12-07:
@zash07 Thank you for your interest in adding this functionality to WEC-Sim, but this is not a feature we plan to implement. If you'd like to run a WEC-Sim case for which you have a known time-series, we recommend converting the wave time-series to the frequency-domain and saving its magnitude a phase. Then you can run WEC-Sim with irregular waves, spectrum import with a seeded phase. Thank you.

closed 2021-12-07 by kmruehl