WEC-Sim/WEC-Sim #772: [BUG] unresolved Link Goto1  in Cable/Base or Follower Drag Body  , MATLAB v2021a

maoyanjun, 2021-11-20:
**Describe the bug**
 when I explored the calbe block in the Model Explorer， It showed Error information for Cabel/Base Drag Body/ Goto1 block: 
```
Failed to find 'WECSim_Lib/Body Elements/Rigid Body/Hydrodynamic Body/Goto1' in library 'WECSim_Lib'.
```
Hence It got multiple errors when running a case of two bodies connecting with calbe like the snipshot below.
**WEC-Sim file(s)**
N/A

**To Reproduce**
N/A

_State method of running WEC-Sim : from command line

**Expected behavior**
I tested several examples, They all worked fine. For now, I just found  the cable block has such problems on my PC. I guess it maybe cause by incompatiblity of MATLAB version.

Besides I want to know how to delete the Base Drag Body and Follower Drag Body in the Cable mask and creat another cable2 mask based on current Cable mask. They are not necessary in my model. 

**Screenshots**
here is the simulink model:
[image]

here is the error messages:
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [ Windows 11]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [ 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [ v4.4]

**Additional context**
N/A

akeeste, 2021-11-24:
@maoyanjun 

Thanks for your question and using the template. This bug is tied to issue #770. There are a few library blocks that were recently renamed and the links in the cable block were broken. This should be a quick fix that I will create a PR for shortly.

On the second half of your question, can you explain what problem the cable drag bodies cause? You can effectively remove these drag bodies by leaving the following variables as the default values:

    cable.bodyMass = 1;
    cable.bodyInertia = [1 1 1];
    cable.viscDrag.characteristicArea = [0 0 0 0 0 0];
    cable.viscDrag.Drag = zeros(6);
    cable.viscDrag.cd = [0 0 0 0 0 0];

If you do this then the cable drag bodies will have a negligible mass and no drag. You could set the bodyMass and bodyInertia even smaller if needed (though they have to be positive and non-zero).Your nonhydro body(3) and body(4) will essentially duplicate any effect you can achieve with the cable drag bodies.

Lastly, I wanted to point out a few issues in the simulink model
- ``constraints(1) and (2)`` are connected to the ``global reference frame`` backwards. We usually recommend that Port B must be connected to the seafloor. This might work but your constraint outputs will always be about port F relative to port B
-  Right now ``cable port B`` is rigidly fixed to the seabed without rotation, and ``cable port F`` will be rigidly fixed to ``body(2)``. Depending on what degrees of freedom ``constraint(2)`` allows, this could work, but it might overconstrain your device motion.

Please let me know if you have any more questions.
Adam

akeeste, 2021-11-29:
@maoyanjun The link issue has been resolved by PR #774. Please pull the most recent version of ``master`` to get this fix.

akeeste, 2021-12-03:
Hi @maoyanjun 

Has this solution and the bug fix in #774 solved your issues? I will close this issue in one week if there are no more questions on this topic.

Thanks,
Adam

kmruehl, 2021-12-07:
@akeeste let's go ahead and close this issue since it's been (hopefully) resolved by #770. @maoyanjun if this issue isn't resolved by #770, please re-open this issue. Thanks!

closed 2021-12-07 by kmruehl