WEC-Sim/WEC-Sim #773: Two Body Point Absorber (RM3)

Devsh93, 2021-11-22:
I keep getting this error, and I don't know how to fix it, any help!?

Error using wecSim (line 46)
['RM3/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 0.012. Stopping
simulation.  There may be a singularity in the solution.  If not, try reducing the step size (either by
reducing the fixed step size or by tightening the error tolerances)
Caused by:
    Error using wecSim (line 46)
    ['RM3/Floating (3DOF)/CONSTRAINT (Planar Joint)']: 'RM3/Floating (3DOF)/CONSTRAINT (Planar Joint)' has a degenerate mass distribution on its follower side.
 
I have tried to change the time step but still get the error.

Thanking you.

akeeste, 2021-12-03:
Hi @Devsh93 

Have any past issues or the troubleshooting page helped resolve this problem? If not please edit this issue to use the Bug Report template and provide additional information.  I will close this issue in one week if there are no further questions.

Thanks,
Adam

Devsh93, 2021-12-04:
Thank you sir.

There is a problem when consider PTO damping coefficient., and also buoy is oscillating not only in heave but also in surge and pitch.

akeeste, 2021-12-06:
Hi @Devsh93 

If you wish to continue discussing this problem or a different aspect of WEC-Sim, please open a new issue using our Bug Report or Theory templates:
[image]

[image]
 
Please fill out all relevant information to clarify the problem and your set-up, and we can assist with your issue.

Thanks,
Adam

kmruehl, 2021-12-07:
@Devsh93 I'm going to close this issue. Please open a new issue using the issue template.

closed 2021-12-07 by kmruehl