WEC-Sim/WEC-Sim #776: User-defined surface elevation and Parallel computing Toolbox

zash07, 2021-11-30:
Hello, 

I have been in contact with David Ogden regarding some errors that I have found and he suggested uploading the issues here. Please refer to the attachment where the issues and the way to solve them are described there.

I was facing some errors when using both user-defined surface elevation and Parallel computing Toolbox. I have tried to check these errors with the final version of WEC-Sim on two different computers. The summary of the errors is given below and can be found in the attachment. 

-	User-defined Surface elevation: When trying to run the codes with the user-defined surface I have faced errors. There were very minor issues in codes and after fixing those, the problem was solved.
-	Parallel Computing Toolbox: I solved the error by writing a code for my research problem. 
Hope you find them useful. 
[attachment](https://docs.google.com/presentation/d/1scklUBzClju3bOabiOCH_gIeACzUcouY/edit?usp=sharing&ouid=111105143475656277241&rtpof=true&sd=true)

Have a nice day. 
Zahra

dav-og, 2021-12-15:
Hi Zahra,
Thanks for raising this - please could you fill in one of our bug report templates (as two separate issues for elevation and pct):
[image]
Cheers,
Dave

zash07, 2021-12-17:
Hi Dave,

I created them, hope you find them useful.

Best regards
Zahra

[image]<https://user-images.githubusercontent.com/12831266/146208029-c89ce0b7-be93-4d7d-8267-0726507ceffb.png>

När du har kontakt med oss på Uppsala universitet med e-post så innebär det att vi behandlar dina personuppgifter. För att läsa mer om hur vi gör det kan du läsa här: http://www.uu.se/om-uu/dataskydd-personuppgifter/

E-mailing Uppsala University means that we will process your personal data. For more information on how this is performed, please read here: http://www.uu.se/en/about-uu/data-protection-policy

dav-og, 2021-12-17:
thanks Zahra! appreciate it

closed 2021-12-17 by dav-og