WEC-Sim/WEC-Sim #778: [BUG] Issue When Implementing The Cable Block

vibishana, 2021-12-05:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
Hello! This issue is connected to #763 where I was trying to model a two-body bottom fixed point absorber with one semi-submerged cylinder and a completely submerged sphere connected by a wire. Previously I was working with MATLAB R2020a and due to compatibility issues (I suppose), the cable block wasn't visible to me. So I used a translational PTO to represent the cable between two bodies (Cylinder and Sphere). Now I have upgraded to MATLAB R2021b and replaced the PTO with a cable block with translational constraints on either side. In #763 I've mentioned that the model has 3DOF although I have considered heave in the present case to simplify the model. Unfortunately, it produces the following errors.

1.	When fixed time steps are used I get the error shown in the screenshot. 

2.	When variable time steps are used simulation works although Cylinder moves independently and the cable does not have any effect on the submerged sphere. Maybe I'm not using the cable block correctly?

**WEC-Sim file(s)**
N/A

**To Reproduce**
1) Run wecSim command as it is. 
2) Change the solver to ode45 and run the same file. (simu.endTime = 50 was used to reduce the computation time)

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
command line

**Expected behavior**
For both bodies to respond in heave 

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
-OS: [Windows 10]
-[MATLAB/Simulink Release][R2021b]
-[WEC-Sim Release][v4.4]

**Additional context**
These are my files 
[POINT_ABSORBER.zip](https://github.com/WEC-Sim/WEC-Sim/files/7655460/POINT_ABSORBER.zip)
Thank You!

dforbush2, 2021-12-16:
Hi @vibishana and thanks for opening your issue.

I'm running 2021a so I needed to rebuild your model based on your description. I was a bit unclear where your PTO went in the cabled model so I used 3 translational constraints, with a cable connecting the 2 bodies of the point absorber. 

I suspect the difficulties you had were in the definition of your bodies/constraint locations and constructing the model in a corresponding way, but I can't be sure since I couldn't open the 2021b version. Runs well for me with the expected cable dynamics in both types of solvers. See the attached model and input file. Please double-check that it is modeled as you have intended, since I was sort of guessing based on your above description.

[revisedModel.zip](https://github.com/WEC-Sim/WEC-Sim/files/7724244/revisedModel.zip)

Generally, if the cable becomes very stiff relative to other components in the system, a smaller time step may need to be specified in the fixed-step solver to retain stability, but I don't think that was the issue you were running into.

Some additional remarks (ignore if irrelevant or unhelpful): 
*the cable specs you gave are pretty unrealistic. I would expect your cable to 
have a large spring stiffness value and a relatively small damping. 
* in regular waves, you can expect after a few full-amplitude cycles to be seeing consistent results. It is probably not necessary to run the simulation for quite so long. 
* verify that your convolution integral time, specified when writing your H5 file is adequately long.

dforbush2, 2021-12-17:
Hi @vibishana 

I verified this on 2021b. Just replace the constraints in my uploaded model with translational PTOs if desired, just make sure to also update your input file as appropraite!

vibishana, 2021-12-18:
Hello @dforbush2

I was able to run it successfully. Thanks for your insight!

closed 2021-12-18 by dforbush2