WEC-Sim/WEC-Sim #781: [User-defined Surface Elevation]

zash07, 2021-12-17:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
_When trying to run the codes with the user-defined surface I have faced errors. There were very minor issues in codes and after fixing those, the problem was solved._

**WEC-Sim file(s)**
_ In the WaveClass.m file, function waveElevUser._ 

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to 'RM3 example'_ 
2. _Click on 'Run for user-defined surface elevation'_ 
3. _Scroll down to 'command window'_ 
4. _See error_

**Expected behavior**
_To be able to use the user-defined surface elevation without error._  

**Screenshots**
_I fixed the error and I do not get the error anymore.  However, you can see the screenshot of how it is fixed below._
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [ 2020b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [Version: 4.3] although with order one also sis not work. 

**Additional context**
_N/A._

dav-og, 2021-12-22:
Hi Zahra - would it be possible to share the input data you're using? It seems like maybe we need a check to see if its column or row vectors - then we can perform the transpose or not, as required.
Cheers,
Dave

zash07, 2022-01-05:
Hi Dave,

Regarding inputs:

1.       The problem with the user-defined surface elevation, I have put an example from your RM3 case (I think you meant this one but I am not sure…). So, you have the possibility to check and see.

2.      The problem with parallel computing, that project is still undergoing, and unfortunately at the moment,  I cannot share.

Best regards
Zahra

När du har kontakt med oss på Uppsala universitet med e-post så innebär det att vi behandlar dina personuppgifter. För att läsa mer om hur vi gör det kan du läsa här: http://www.uu.se/om-uu/dataskydd-personuppgifter/

E-mailing Uppsala University means that we will process your personal data. For more information on how this is performed, please read here: http://www.uu.se/en/about-uu/data-protection-policy

akeeste, 2022-01-05:
Hi @zash07 

I recently ran our RM3 example using the [etaData.mat](https://github.com/WEC-Sim/WEC-Sim/blob/master/examples/RM3/etaData.mat) user defined wave elevation data. I did not encounter a bug running WEC-Sim from either the command line or from Simulink. v4.3 on R2020b and the master branch on R2021b are both working.

Can you share a sample of the input data for this case? There is not much I can do without recreating the bug. 

From your screenshots I am guessing that your input data is transposed incorrectly.  The data should be one array, size 
``[# time steps, 2]``, where the first column is time and the second column is the wave elevation at the origin

Adam

zash07, 2022-01-07:
Hi,

I am quite preoccupied at the moment and in the following months, but I have checked one more time for my script and removed the edited version that I created (so I use the original WECSIM files).  The following error is given:

***@***.***

Unfortunately, I am not able to share my input file due to confidentiality reasons, but the format of my file is as below:
***@***.***

I have edited the waveClass.m as I mentioned previously both in my emails and in the case that I created in your GitHub. Then again, I checked both my script (wecSim File) and your RM3 case and I see that now both are working just fine.

Here, I have mentioned all I knew and did.

\Zahra

akeeste, 2022-01-10:
Hi @zash07 

Unfortunately it seems the additional information you provided was removed by the email server:
> The following error is given: ***@***.*** Unfortunately, I am not able to share my input file due to confidentiality reasons, but the format of my file is as below: ***@***.***

I'm glad your solution has fixed your problem. I suggest transposing the input data if it occurs again. Since we cannot recreate this issue and it appears our current versions are working, there is nothing we can do at this time. I will close this issue and we can revisit this if required.

Thanks,
Adam

closed 2022-01-10 by akeeste