WEC-Sim/WEC-Sim #782: [Parallel Computing Toolbox (PCT)]

zash07, 2021-12-17:
**Note: italicized text below is included as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
_The parallel computing (PCT), did not work for me even for the RM3 example._

**WEC-Sim file(s)**
_N/A_

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to 'Rm3 example'_
2. _Click on 'to run pct '_
3. _Scroll down to 'command window'_
4. _See error_

**Expected behavior**
_To be able to work with PCT._

**Screenshots**
_I provide an error for my case the same error can be seen for RM3 original example._
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2020b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [Version: 4.3]

**Additional context**
_I wrote the following scripts for parallel computing to run for different sea states and I could solve my problem following that._
[image]
 
[image]

dav-og, 2021-12-22:
Hoping @jleonqu can help out with this when he's back

jleonqu, 2022-01-11:
Hello @zash07 

I would say the problem is that the information needed to run the simulation is not available for all the workers created with the parfor command. There are some specific restrictions when running simulink models within parfor. In the current documentation in the Mathworks webpage there is an explicit recommendation for not using the sim function within parfor, see this [link](https://www.mathworks.com/help/simulink/ug/not-recommended-using-sim-function-within-parfor.html). The other option is to use parsim, which could be the solution you are looking for.

Another option to solve this issue is to use the Sensitivity Analyzer tool in Simulink (documentation [here](https://www.mathworks.com/help/sldo/ref/sensitivityanalyzer-app.html) ) and the Sensitivity Analysis options in Matlab (documentation [here](https://www.mathworks.com/help/sldo/sensitivity-analysis.html) ). I have used these options in the past for some other projects and it works well. My recommendation for you is to use these options when running a large amount of cases.

jleonqu, 2022-02-02:
Hello @zash07 
Does this answer help?

zash07, 2022-02-03:
Hello,

I sorted my problem in another way as I mentioned it.  Therefore, I just wanted to inform you about the problem. I do not have time to check other solutions at the moment. However, if they are reliable, I suggest that to be included in the manual description, it would be great for others with the same problem as I faced.

Best regards

När du har kontakt med oss på Uppsala universitet med e-post så innebär det att vi behandlar dina personuppgifter. För att läsa mer om hur vi gör det kan du läsa här: http://www.uu.se/om-uu/dataskydd-personuppgifter/

E-mailing Uppsala University means that we will process your personal data. For more information on how this is performed, please read here: http://www.uu.se/en/about-uu/data-protection-policy

jleonqu, 2022-02-08:
Thanks for your answer. I'll close this issue, but feel free to ask more questions or open a new one if needed.

kmruehl, 2022-03-10:
from @yuyihsiang 

Hi All,

I noticed that there was a discussion on the issue #782 regarding wecSimPCT. I ran into the same issue when running some WEC-Sim cases myself.

The issue is that the way the code was written forces the wecSim run to be executed inside the computing working directory, which is generated when calling the parfor. In the past we had a few lines of the code inside wecSim.m file to change the directory back to the main working directly before calling the sim function and move back inside the computing working directory after the sim function is done. However, that part of the code does not exist anymore. I am note sure if there is a reason why it was removed. I think we should try to put it back if possible.

Best,

--Yi-Hsiang

kmruehl, 2022-03-10:
I have also run into this issue

```matlab

Simulating the WEC device defined in the SimMechanics model C:\Users\kmruehl\Documents\GitHub\WEC-Sim\WEC-Sim_Applications\Multiple_Condition_Runs\RM3_MCROPT1\RM3.slx...   
Elapsed time is 9.171886 seconds.
Error using wecSim (line 46)
No system or file called 'RM3.slx' found.

Error in wecSimFcn (line 23)
wecSim

Error in wecSimPCT (line 102)
parfor imcr=1:length(mcr.cases(:,1))

```

kmruehl, 2022-03-17:
@yuyihsiang  this issue has been resolved, thanks @akeeste for the help!

closed 2022-03-17 by kmruehl