WEC-Sim/WEC-Sim #783: Modeling Two Body Point Absorber in Wec-Sim: Nemoh/Bemio Results Issues

ccz359, 2022-01-11:
I’m working on modeling a wave powered water pump using Wec-Sim. I’ve followed along with the online tutorials and manuals and have successfully run the RM3 Wec-Sim example. I’ve started applying the water pump design to the RM3 example and am stuck on the BEM software step. The free version of WAMIT does not work for two body systems, so I’ve been using Nemoh. I am using the master branch version of Nemoh from GitHub.

I’ve gotten Nemoh to run with the water pump design. However, when looking at the Nemoh/Bemio tutorial page (https://wec-sim.github.io/bemio/nemoh.html), I noticed that my mesh output file names do not match. When I run Nemoh, there are no Hydrostatics or KH.dat files are generated within the Mesh folder. Instead, L10, L12 and Integration files are generated. For other Nemoh examples, this is also the case.

Is there another step I’m missing that converts the L10, L12, and Integration.dat files into KH and Hydrostatic files?

Also, there are many empty or NaN results within the Nemoh Results folder when I run Nemoh. Could this be due to the above meshing error? Or is there an issue with the geometry file inputs? I took external surface .stl files from SolidWorks and converted them into .dat files using the Meshmagick application. The .stl files had open faces at the waterline.

Attached is my working folder in Nemoh. Any advice on this is very welcome and appreciated.
[WP.zip](https://github.com/WEC-Sim/WEC-Sim/files/7848234/WP.zip)

kmruehl, 2022-01-11:
Hi @ccz359 Thank for your question and interest in WEC-Sim. While this is a Nemoh question, and the WEC-Sim team does not manage or support Nemoh, we can provide some guidance. Unfortunately the short answer is that Nemoh does not automatically generate a hydrostatic `*KH.dat` file. This file is only generated by the Nemoh meshing scripts, which must be run separately. This is a different workflow than the one you've used converting an STl to a DAT using Meshmagick. However, you can manually calculate the hydrostatics for your device, and generate your own `*KH.dat` file using the WEC-Sim provided example as a template. 

If you're looking for another free alternative or an open-source BEM solver, WEC-Sim is also compatible with Capytaine.

ccz359, 2022-01-11:
Hi @kmruehl 

Thank you for your quick and helpful response! There were a few items I should add, I did reach out to the Nemoh developer team last week with this question but have not heard back from them yet. 

I also did try your suggestion of manually calculating the KH and Hydrostatic files for my design, using the RM3 as an example template. The Results still had the NaN issues mentioned above. 

I will try Capytaine and see how that goes.

Thanks again!

waveenergyharnessingdevice, 2022-01-11:
As in the “Shoaling Feature” of US Patent 8093736B2?

Any “bluff body” creating a high rate of sheer in a flow field creates a “Shoaling Feature” as a high sheer rate for water flow in waves causes viscosity to dominate the flow field and the “Divergence Operator” to become a negative value or non-zero making the “inviscid” and “0” value for the Divergence Operator assumptions commonly used in Airy and Navier-Stokes wave equations not valid.

Please note this for DOE funded work.

Best regards,
Michael Raftery

kmruehl, 2022-01-12:
Hi @ccz359, thank you for the update. We're aware of the lack of Nemoh support, and are actively working to expand WEC-Sim integration with other open-source BEM software (like Capytaine), and our long-term plan to support an open-source BEM as part of the WEC-Sim project (in the future).

"I also did try your suggestion of manually calculating the KH and Hydrostatic files for my design, using the RM3 as an example template. The Results still had the NaN issues mentioned above." That's very strange, hopefully Capytaine will work for you. I'm going to close this issue, but Capytaine is being supported.

closed 2022-01-12 by kmruehl

pn-206, 2022-02-04:
Hi @ccz359 

I faced exactly the same issue and found a workaround which I've described below.  I hope it helps you and and anybody else who finds themselves in a similar position.

Firstly, as @kmruehl has already mentioned, the KH.dat and Hydrostatics.dat file are not created by running the main Nemoh process but by running a separate meshing process.  The required file, Mesh.exe can be found be found by downloading the "Matlab routines for Nemoh" from [here](https://lheea.ec-nantes.fr/valorisation/logiciels-et-brevets/nemoh-installation).

What's really confusing, is the that mesh file that Mesh.exe requires, is not the same mesh file as required by the primary Nemoh processes.  I used [BEMRosetta](https://github.com/BEMRosetta/BEMRosetta) to create these different formats.  You can import STL files into it and save them as "Nemoh .dat" files and "Nemoh premesh" files.  The premesh file is what you'll need to run Mesh.exe; instructions can be found [here](https://lheea.ec-nantes.fr/valorisation/logiciels-et-brevets/nemoh-mesh).

The outputs of successfully running Mesh.exe are the generation of a mesh file for Nemoh (which you shouldn't need if you used BEMRosetta) but also the KH.dat and Hydrostatics.dat files which you will need for BEMIO.

Hope this helps.