WEC-Sim/WEC-Sim #784: Does the PTO-Sim block can accept user defined forces/values in hopes of providing real-time control?

FFxiaolang, 2022-01-13:


akeeste, 2022-01-13:
Hi @FFxiaolang 

Thank you for your question. We require users to open GitHub issues using an appropriate template. This is useful to the development team to better classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please edit this issue to use the "Theory or Implementation" template and link to the relevant documentation that is unclear, I am happy to help.

Thanks!
Adam

akeeste, 2022-01-25:
Hi @FFxiaolang 

Since I have not heard back in two weeks I will close this issue. Please reopen your question using a provided template if further clarifications are needed. 

I will also refer you to our documentation on [PTO Advanced Features](http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#constraint-and-pto-features) and [PTO blocks](http://wec-sim.github.io/WEC-Sim/master/user/code_structure.html#pto-blocks). These pages should help answer your question and clarify how to use these blocks. 

Thanks,
Adam

closed 2022-01-25 by akeeste