WEC-Sim/WEC-Sim #785: PTO-Sim linear/rotary generator, guidance to choosing parameters

t-osu, 2022-01-13:
**Is your question request related to a problem? Please describe.**
_N/A_

**Describe the theory or implementation approach that you have a question**
Hello all, I have been reviewing how to use the rotary generator in PTO-Sim, however the only example WEC-Sim provides is for the RM3 with the linear generator. The OSWEC uses hydraulics then the lookup table. I switched the OSWEC hydraulics to the rotary generator and the simulation was unsuccessful; I figure because the parameters in the _ptoSimInputFile.m_ for the linear generator were specific to the RM3. All I did was copy and paste these parameters for the rotary generator (in addition to adding the rotor radius).

I am looking for guidance on how to implement this for the OSWEC to run successfully, which probably needs different motor parameters. Does one choose motor size solely on how much power you want out? The sea state? How would you size a motor for your system? Can it be too big and interfere with the dynamics? Also, when reviewing manufacturer spec sheets, the more specific parameters such as magnetic pole pitch or inductance of the coil are not given that I have found. 
I would be very appreciative of any resources/references I could read up on how to begin sizing the rotary generator block to a system. I read issue #550, but it didn't quite clarify my thoughts.

**Personal project context**
I have been working on a vertical axis pendulum PTO for a WEC and would now like to replace the ideal rotational PTO with a more realistic PTO.

**Describe the type of conclusion or resolution you are looking for**

**Additional context**
_N/A_

Thank you, please let me know if I need to clarify anything or if it needs to be moved to a different discussion topic,

jleonqu, 2022-01-19:
Hello @t-osu

You need to choose your motor based on the power rate you need, the max. shaft speed, and the max. torque. You can estimate these parameters based on your inputs, which is this case could be the sea state and the mass of your system. Once you have all the characteristics of your motor (power, speed, and torque) you can try to look online for one specific motor with these characteristics, and then in the datasheet you can find some of the parameters that you can use for your model, such as the armature resistance, the armature inductance, the back emf constant, etc.

t-osu, 2022-01-22:
> [This email originated from outside of OSU. Use caution with links and
> attachments.]

Morning,

Thank you, that helps a little. Could you please elaborate more on how to
estimate the power rate/shaft speed/etc parameters....based off of sea
state and mass of the system?
The only power I can estimate from the sea state that I know of is using
assumed deep depth energy equation (J = pg^2 (0.85Tp Hs^2)/64*pi) , and
then getting power from the empirical study by Babarit where they estimate
efficiency for certain WEC topologies yielding power = J*B*efficiency. This
uses the characteristic dimension though and not the mass.

So I'm still lost on which equations that you are referring to in order to
obtain the motor parameters you spoke of. A quick example would be very
beneficial, perhaps using the RM3 DD PTO-sim example? The sea state in the
RM3 example with the linear motor was T = 8s H = 2.5m and the mass was
'equilibrium' for both. I guess since the mass was assumed to be that of
the water volume, lets just say the spar was 10kg and the float 5 kg (yes
very small, but just for the purpose of explaining).
 How would one proceed from this information to estimate power rate/shaft
speed/torque? Or is there a paper I can refer to?

Once I know how to estimate those, then yes, finding an online motor should
be doable.
Thank you.

jleonqu, 2022-01-25:
If I understand correctly, what you want is to implement in the OSWEC example something similar to the RM3 direct drive, so it is like a rotary generator directly connected to the rotational join of the OSWEC. What I recommend is to run the OSWEC example just as it is with the lookup table, then you can see the results for torque and rotational speed on the rotational PTO:

[image]

This block represents the rotational join of the OSWEC. If my understanding is correct, here is where you want to put your motor. As I mentioned, you need to check the results for torque and rotational speed in this join. After you run the WEC-Sim model, you can check the torque and the speed with the following commands:

torque = output.ptos.forceTotal(:,5);
Speed = output.ptos.velocity(:,5);

With this information you can proceed and select an electric motor.

t-osu, 2022-01-26:
Hi Jorge,

Yes, that is how I have it set up, except I was having issues defining the
parameters you need for the rotational generator block.
Actually, since you do not have the parameters for it, you would still be
using only the rotational PTO and the torque and speed that it sees through
simulations and find a motor that fits those values, as you pointed out.
Much simpler than I was making it, thank you for the chat!

closed 2022-01-31 by jleonqu