WEC-Sim/WEC-Sim #786: Visualization Issue (RM3)

BHShanab, 2022-01-21:
Hi Everyone,
I'm a new user of WEC-SIM. I'm doing a simulation on designing the second body system of the Two-body WEC System. I use a specific buoy shape as a float and designed two different shapes of the second body so far, (plate & cylinder). I got some issues when running wecSim, if you could help me to tackle these issues below:
(1) I use AQWA, results in the frequency domain from AQWA are reasonable now. However, I found that visualization of the system via WECSIM does not work as I designed in AQWA in both systems as attached; (two bodies are not on the same axis and both are not in the designed coordinates). so, if you could tell me how can I identify the problem and solve it. 
(2) Does it affect my results?
(3) I see also from the visualization that wave motion and the body are not in the heave motion only as I need like the original RM3 example, and for the cylinder body, it is not stable, it rotates at a specific time!!
Thanks in advance for your kind help and support!
Regards!
Belal 
[cylinder.zip](https://github.com/WEC-Sim/WEC-Sim/files/7909704/cylinder.zip)
[Plate.zip](https://github.com/WEC-Sim/WEC-Sim/files/7909705/Plate.zip)
[image]

nathanmtom, 2022-01-24:
@BHShanab Thank you for your inquiry and I'd be happy to help you, but we are asking users to submit issues using one of our templates. 

If you go to New Issue, click `WEC-Sim applications specific issue or question' Getting Started tab (as you appear to be working off of the RM3 model. ), and please complete all sections of the template but do not delete the bolded headers. We have example language to use in italics to help you describe your issue. 

Once you edit this issue to follow the template or open a new issue I will be glad to help troubleshoot your issue.

[image]

nathanmtom, 2022-01-24:
@BHShanab Thank you for updating the issues template, I am going to close this issue and answer your questions in the recently opened issue.

closed 2022-01-24 by nathanmtom