WEC-Sim/WEC-Sim #787: [How can I adjust the Visualization part in (RM3) model?]

BHShanab, 2022-01-24:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**
I'm a new user of WEC-SIM. I'm doing a simulation on designing the second body system of the Two-body WEC System. I use a specific buoy shape as a float and designed two different shapes of the second body so far, (plate & cylinder). I got some issues when running wecSim, if you could help me to tackle these issues below:

**To Reproduce**
(1) when I use wecSim command in MATLAB after completing the required steps before this command, I see an error in the visualization part.
 I use AQWA, results in the frequency domain from AQWA are reasonable now. However, I found that visualization of the system via WECSIM does not work as I designed in AQWA in both systems as attached; (two bodies are not on the same axis and both are not in the designed coordinates). so, if you could tell me how can I identify the problem and solve it.
(2) Does it affect my results?
(3) How can I control the heave motion response only?
I see also from the visualization that wave motion and the body are not in the heave motion only as I need like the original RM3 example, and for the cylinder body, it is not stable, it rotates at a specific time!!

**Expected behavior**
I expect the get the correct oscillation in the heave motion on the same axis as modeled.

**Screenshots**
[Plate.zip](https://github.com/WEC-Sim/WEC-Sim/files/7929545/Plate.zip)
[cylinder.zip](https://github.com/WEC-Sim/WEC-Sim/files/7929546/cylinder.zip)
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 11]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2021a], 
 - I updated MATLAB two days ago to be 2022a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [Version: 4.4]

**Additional context**
Thanks in advance for your kind help and support!
Regards!
Belal

nathanmtom, 2022-01-24:
@BHShanab Thank you for opening a new issue with the required template completed. 

First off, it looks like at time t=0 you have the right orientation for your `.stl` files, see below:

[image]

However, after a few seconds your model appears to rotate 90 degrees about the y-axis, see below:

[image]

This is likely because your center of gravity and the center of buoyancy of body(1) are not coincident, see below. Therefore, without additional restoring or external forces the body will rotate as shown in the above screenshot.

[image]

Therefore you will need to check your inputs into AQWA to ensure that you have placed the mesh of the top body in the correct location. 

Lastly, the scale is off in the visualization because in your `.stl` file it appears you exported in millimeters rather than meters which is required by WEC-Sim. Please double check when exporting you have the correct units defined.

I hope these responses point you in the right direction for trouble shooting your WEC-Sim model. 

Please let me know if you need any of the above points clarified or you have additional questions.

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2022-02-02:
@BHShanab I wanted to check to see if you were able to resolve your issue above? If we do not hear from you in the next week we will close this issue. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

BHShanab, 2022-02-02:
I'm still working on it. I will update you once I finish.
Thanks!

nathanmtom, 2022-02-16:
@BHShanab Since we have not heard from you in two weeks, we are going to close this issue. Please re-open this issue if you have still not resolved the problems described in this issue or open a new issue if you have resolved these issues but have new issues.

Cheers, 
Nathan
National Renewable Energy Laboratory

closed 2022-02-16 by nathanmtom