WEC-Sim/WEC-Sim #788: MoorDyn use in MATLAB

meganandd, 2022-01-26:
**Is your question request related to a problem? Please describe.**
My question is related to MoorDyn and loading the library in MATLAB. I am trying to use the loadlibrary function in MATLAB, but it seems like this is for C rather than C++. However, I copied the sample code from the MoorDyn documentation published by Matt Hall in 2015 and am still getting errors. It might be that it is just outdated, but I feel like I am missing something. I'll attach an image of my code and then also the error message I am getting in the console. 

**Describe the theory or implementation approach that you have a question**
N/A

**Personal project context**
I am a student working on a mooring simulation for a rotating mass WEC. 

**Describe the type of conclusion or resolution you are looking for**
I would like some guidance on loading the library into MATLAB. 

**Additional context**
[image]
[image]

akeeste, 2022-01-26:
Hello @meganandd 

Thank you for opening an issue by completing the template. 

I am not familiar with the sample code given in the MoorDyn documentation, but it seems like the ``Lines.dll`` file is missing. Note that the [loadlibrary](https://www.mathworks.com/help/matlab/ref/loadlibrary.html#btgww7p-1) function is taking two inputs: "Lines" which is the name of the dll file, and "MoorDyn" which is the name of the header file.

Also, right now WEC-Sim only supports a specific version of MoorDyn so you might come across other issues with a different version. Please download the version at [WEC-Sim/MoorDyn](https://github.com/WEC-Sim/MoorDyn) and unzip into the ``WEC-Sim/source/functions/moorDyn`` directory. This repo contains the Lines library for different operating systems and the required header file.

Please let me know if there are additional issues getting MoorDyn installed.

Adam

akeeste, 2022-02-08:
Hi @meganandd 

One of my colleagues pointed out that there is a typo on Line 12 of your pasted code, this may be the root of your issue.

I have not seen any movement here for two weeks so I will close this issue. If there are further questions regarding the WEC-Sim/MoorDyn coupling please open a new issue to address this.

Adam

closed 2022-02-08 by akeeste