WEC-Sim/WEC-Sim #792: [BUG] OSWEC Modeling

vibishana, 2022-01-30:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
Hello! I'm modeling an OSWEC with one of my own geometry. Everything works well for regular waves although for irregular waves following error occurs even for very small time-steps. Any help would be much appreciated.
Thank you!

**WEC-Sim file(s)**
[OS_WEC.zip](https://github.com/WEC-Sim/WEC-Sim/files/7965615/OS_WEC.zip)

**To Reproduce**
Run wecSim

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
command line

**Expected behavior**
N/A

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - [MATLAB/Simulink Release] [R2021b]
 - [WEC-Sim Release][v4.4]

**Additional context**
N/A

akeeste, 2022-01-31:
Hello @vibishana 

Thank you for using the template and uploading your case file. 

I checked your case and found that the ``noWaveCIC``, ``regularCIC`` and state space wave cases also fail. This indicates that there is an issue with your BEM data and the IRFs created with BEMIO. Can you also upload your BEMIO input so that I can view it? 

To solve this problem, I highly recommend that you:
- plot the BEMIO output with ``Plot_BEMIO`` and check that the conditions list on the plots are being met (damping decays to zero, added mass decays to constant, IRFs decay to zero, etc)
- perform each [numerical test case](http://wec-sim.github.io/WEC-Sim/master/user/troubleshooting.html#numerical-test-cases) whenever a new WEC-Sim case is developed

Please let me know what information these steps provide.
Adam

vibishana, 2022-02-01:
Hello!

 Yes, you were right about the BEMIO output. I found out that conditions are not met in added mass and radiation damping.
Here are my files,
[oswec.zip](https://github.com/WEC-Sim/WEC-Sim/files/7977767/oswec.zip)

Does it mean I have to change my geometry or is there another solution? 

Thank you!

akeeste, 2022-02-01:
Hi @vibishana 

I am not very familiar with Nemoh, so it is difficult to give a single solution. It looks like the BEM meshes have 1602 and 536 panels respectively and run at 25 frequencies. I recommend a high resolution mesh, and definitely recommend a higher resolution frequency range. For WEC-Sim, the BEM should use at least 200 frequencies or more.

Unfortunately this is all the support we can give on using BEM codes. I am happy to provide more assistance if there is an issue with WEC-Sim or BEMIO.

Thanks,
Adam

vibishana, 2022-02-03:
Hello @akeeste 

I was able to run the simulations without any issues following your recommendations. Appreciate the support.

akeeste, 2022-02-08:
@vibishana Good to hear that this helped resolve your problem. I will close this issue

closed 2022-02-08 by akeeste