WEC-Sim/WEC-Sim #793: [BUG] Utilizing a fixed-step time solver when simu.dt and simu.dtOut are not equal

nathanmtom, 2022-01-31:
**Describe the bug**
There have been several recorded instances where the error show in the screenshots section occurs when using a fixed step time solver `ode4` while setting `simu.dt` and `simu.dtOut` to different values. This has so far not been an issue when using the variable time step solver `ode45`.

Anticipate that changes to the rate transition block made to ensure consistent time stamps for the variable time step has caused the issue. Past models with a fixed time step and different `simu.dt` and `simu.dtOut` were working as expected.

**WEC-Sim file(s)**
Currently, the issue is likely to be caused by the rate transition block in the `to workspace` block that outputs recorded quantities of interest at a different sampling rate than the simulation time step.

**To Reproduce**
_Steps to reproduce the behavior:_

1. Open example RM3 model.
2. Add simu.dtOut = 0.2.
3. Type WEC-Sim in the command line.
4. See the following error shown below in screenshots.

**Expected behavior**
Regardless if a fixed or variable time step solver is used, the simulation time step and the output time step are different the simulation should execute successfully.

**Screenshots**

[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB/Simulink Release:  2021a
 - WEC-Sim Release: v4.4

**Additional context**
N/A

akeeste, 2022-02-01:
Hi @nathanmtom 

I looked into this issue a bit today. There are two solutions that I found:
- explicitly add a rate transition block immediately before the ``To Workspace`` block
[image]

- change the ``To Workspace`` signal from being a `bus` to a vector (`mux` block)
[image]

Interestingly, this error does not occur with our PTOs or Constraints, which use an identical ``To Workspace`` block as the body library. The difference between them is that the PTOs/Constraints uses a ``mux`` block to combine variables for output, whereas the body uses a ``bus`` to combine variables for output. The differences between the two are explained [here](https://www.mathworks.com/help/simulink/ug/composite-signal-techniques.html#bvevlbu). They should both be [virtual signals](https://www.mathworks.com/help/simulink/ug/nonvirtual-and-virtual-blocks.html) so I'm not completely sure why the mux works but the bus doesn't. 

The ``To Workspace`` block suggests that bus signals use a Timeseries output format. This might be why the ``bus`` fails when writing to structure output, but `mux` does not. Writing either to a Timeseries output also prevents the above error, but this [option does not support Rapid-accelerator mode](https://www.mathworks.com/help/simulink/slref/toworkspace.html)
[image]

Lastly, [this page ](https://www.mathworks.com/help/simulink/ug/resolving-rate-transitions.html )might provide more insight into why the variable-step and fixed-step solvers behave differently. It seems they could have different levels of rate transition handling, perhaps why the fixed-step fails and the variable-step does not.

Adam

nathanmtom, 2022-03-10:
Development within PR #799 should resolve this issue.

closed 2022-03-15 by kmruehl

akeeste, 2022-08-12:
Noting that this bug has been resolved by the latest MATLAB update (R2021b update 4). See follow-up from Mathworks below:

    Dear -----------
    
    We are pleased to inform you that the issue you reported has been addressed in the latest R2021b Update release. Please see below for more information about this issue. 
    
    Our records indicate that on February 25, 2022, you contacted technical support regarding Simulink 10.4 (R2021b). The topic of the issue was: Simulink solver cause strange low-frequency oscillations
    
    You can find out more information about this issue here:
    http://www.mathworks.com/support/bugreports/2693577?ref=ts_R2022a_Update_4