WEC-Sim/WEC-Sim #794: [Theory or Implementation]

Dadidal, 2022-02-01:
**Is your question request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I do not understand why I get 'X' result when running 'Y' simulation [...]_

**Describe the theory or implementation approach that you have a question**
Hello! I am trying to simulate simple ship-wave interaction using WecSim. Firstly I used NEMOH as BEM solver and created .h5 file based on NEMOH results. Everything worked well but recently I understood that **I need to use Nonlinear Buoyancy and Froude-Krylov Excitation** (https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#nonlinear-buoyancy-and-froude-krylov-excitation)

I added 2 parameters to my input file:
simu.dtNL=1*simu.dt;
body(1).nlHydro = 2;

After trying to solve my problem I got the following **Error:**
Error using wecSim (line 46)
Invalid setting in block 'AE/Float/Hydrodynamic Body/Wave Diffraction and Excitation Force Calculation/Nonlinear FK Force Variant Subsystem/With
Nonlinear FroudeKrylov Force/Deep Water Flag' for parameter 'Value'

[image]

I checked the documentation and observed that **waves.deepWaterWave** property is read-only and WecSim read it from NEMOH.
Can you please explain me how to solve this issue? I cheked NEMOH documentation and didn't find any related parameters.

I attached my project archive.
[AE.zip](https://github.com/WEC-Sim/WEC-Sim/files/7975167/AE.zip)

akeeste, 2022-02-01:
Hi @Dadidal 

Thanks for bringing up this issue. The problem is actually related to the definition of ``waves.waterDepth``. This parameter should only be specified in the input file if no hydro bodies are being used, otherwise it is read from the .h5 file. When it is specified ahead of time, the ``waves.deepWaterWave`` flag is not set properly. It remains empty and gives that error in Simulink. Removing the definition of ``waterDepth`` and ``deepWaterWave`` from the input will resolve this issue.

@jtgrasb Let's add a warning flag that will catch if ``waves.waterDepth`` is defined when there is an .h5 file present.

@Dadidal You mentioned above that you are using NEMOH BEM data. I could not view this as only the WAMIT and AQWA output is uploaded, but I noticed that AQWA and WAMIT both use only 5 frequencies. I caution that the case now runs, but your results may be inaccurate unless more frequencies are used. I recommend reducing the number of panels in the mesh and using ~200+ frequencies

Please let me know if you have any more questions.
Adam

Dadidal, 2022-02-06:
Hello @akeeste! Thank you very much for your help and your recommendation, the problem was solved!

closed 2022-02-06 by Dadidal