WEC-Sim/WEC-Sim #795: ptoSimInputFile for Linear Generator

aglassFHKiel, 2022-02-02:
Hey,
By modifying the ptoSimInputFile, some coefficient inputs are unclear to me.
Rs-Winding resistance: is it the resistance per coil oder per winding?
Bfric-Friction coefficient: what friction is this ment to be an how is it calculated in this example?
lambda_fd- Flux linkage: how is it calculated? 
Is there an Documantation where i can look it up?
Thanks in advance for your help!

[image]

jleonqu, 2022-02-02:
Hello @aglassFHKiel 

Rs is the resistence per winding.
Bfric is the friction coefficient that account for the friction losses in the electric generator.
lambda_fd: this is the flux linkage of the stator d winding. 

Bfric and lambda_fd are not calculated, these values are properties of the electric generator used as a case study for this example.

Regarding documentation, please look at references 7 and 10 in this link: [WEC-Sim Publications](http://wec-sim.github.io/WEC-Sim/master/introduction/publications.html)

jleonqu, 2022-02-16:
I'll close this issue, but feel free to ask more questions or open a new issue if needed.

closed 2022-02-16 by jleonqu