WEC-Sim/WEC-Sim #798: [BUG] Sigularity  Issue

FFxiaolang, 2022-02-07:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
I was very interested in Dr. Ransco's  Control-Sim project .I refer to @ratanakso's [PhD dissertation](https://ir.library.oregonstate.edu/concern/graduate_thesis_or_dissertations/z316q655j). When I run this project, I can't seem to debug this one. The error is:
  
[image]
    I changed simu.dt from 0.1 to 0.01 or 0.001. I've tried changing the wave type(irregular or regular) and changing the simu.solver to ode4 and ode45. But none of these fix the error.
**WEC-Sim file(s)**
[Control-Sim-master .zip](https://github.com/WEC-Sim/WEC-Sim/files/8013364/Control-Sim-master.zip)

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to '...'_
2. _Click on '....'_
3. _Scroll down to '....'_
4. _See error_

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2020b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v4.4]

**Additional context**
_Add any other context about the problem here._

FFxiaolang, 2022-02-07:
[Control-Sim.zip](https://github.com/WEC-Sim/WEC-Sim/files/8015973/Control-Sim.zip)

dforbush2, 2022-02-16:
These changes have been pushed. Please open a new issue if you encounter additional difficulty!

closed 2022-02-16 by dforbush2