WEC-Sim/WEC-Sim #805: [WEC-Sim Applications]- Global Coordinate System

aristi-pap, 2022-02-21:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
[WEC-Sim_Applications](https://github.com/WEC-Sim/WEC-Sim_Applications)/Nonlinear_Hydro/

**Describe the bug or issue**
I have a question about the global coordinate system in WEC-Sim and the positions of the constraints and PTOs. The manual states that the Global Reference Frame block, which acts as the seabed, defines the global coordinates.  In the wecSimInputFile the locations of the constraint and PTOs are defined by:
constraint(i).loc = [<x> <y> <z>];
pto(i).loc = [<x> <y> <z>];
My hydrodynamic data are generated using the boundary element method (BEM) solver WAMIT. In WAMIT, the origin of the global coordinate system is located on the free surface.
According to the Nonlinear Buoyancy and Froude-Krylov Wave Excitation Tutorial - Heaving Ellipsoid, the water depth is 70m and the position of the constraint(1) and pto(1).loc is [0 0 -12.5].  It comes to my understanding that these coordinates are referring to the global coordinate system on the free surface, such as the one in WAMIT, because the coordinate z is a negative number. Shouldn’t the locations of the constraint and PTOs be defined with respect to the global reference frame, in the seabed, and in this case be equal to [0 0 57.5]?

**To Reproduce**
Ν/Α

**Expected behavior**
Ν/Α

**Screenshots**
[image]
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html)  2021a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v4.4

**Additional context**
I have noticed this issue in other examples too, such as the OSWEC, and it would be very helpful if you could explain to me how is the global coordinate system defined in WEC-Sim?

akeeste, 2022-02-22:
Hi @aristi-pap,

Thanks for opening an issue and using the template. I will do my best to address your question.

The short answer is that in WEC-Sim all body centers of gravity and pto, constraint, and mooring locations are given relative to the global coordinate system. The [WEC-Sim coordinate system ](https://wec-sim.github.io/WEC-Sim/master/theory/theory.html#coordinate-system) is defined at the still water line. So the locations given in the applications and tutorials are correct and reference the still water line.

The longer answer is that the ``Global Reference Frame`` block has several uses within WEC-Sim. The broad purpose of the ``Global Reference Frame`` is described in the [Code Structure documentation](https://wec-sim.github.io/WEC-Sim/master/user/code_structure.html?highlight=global%20reference%20frame#frame-block). It determines the solver configuration, global settings, the global coordinate system, the mean free surface, and the seabed. This is seen under the block mask:
[image]

The global reference frame defines the coordinate system described in the [theory section ](https://wec-sim.github.io/WEC-Sim/master/theory/theory.html#coordinate-system). But it also defines the world frame and seabed as rigidly fixed, immovable frames to which everything else in the model is connected. The wording in the [tutorial documentation](https://wec-sim.github.io/WEC-Sim/master/user/tutorials.html?highlight=global%20reference%20frame#step-2-build-simulink-model) is meant to highlight that the ``Global Reference Frame`` represents the immovable mass of the seabed in a physical system.

Let me know if you have more questions on this. I will start a PR to update the documentation and clarify these distinct purposes of the ``Global Reference Frame``

Adam

aristi-pap, 2022-02-27:
Hello @akeeste! Thank you very much for your help and for your quick response.

closed 2022-03-10 by nathanmtom