WEC-Sim/WEC-Sim #810: Question regarding the explanation of this shift of the WEC

eleftheria-pap, 2022-02-24:
Hello, 

**Is your question request related to a problem? Please describe.**
N/A

**Personal project context**
I am developing a model with a WEC in front of a fixed wall. The WEC has only heaving motion as the remaining degrees of freedom are constrained and the wall has all degrees of freedom constrained as a fixed body. The wave direction is shown below in the attached image. 

**Describe the theory or implementation approach that you have a question**
I have a question regarding the results of the implementation. At the beginning of the simulation there is a small shift of the position of the WEC towards negative values and then the simulation continues as expected.

**Describe the type of conclusion or resolution you are looking for**
Can you explain why this shift is happening ? If this is not an error that may occured, what is the physics behind this shift?

**Additional context**
Wave direction
[image]

The graph that presents the shift of the WEC in the beginning (below the free surface)
[image]

nathanmtom, 2022-02-25:
@eleftheria-pap Thank you for submitting your question about your WEC-Sim model and I will do my best to answer your question. 

First off, it is difficult to tell from just the time history of the heave response if there is any potential issue you are having, but these are my suggestions and thoughts on how to troubleshoot this issue. 

- I'd like to see a `noWaveCIC` case to see if you still have a drop in the heave position of the WEC. If so then this becomes a mass and hydrostatic problem and you should check your mass properties.
- In your wecSimInputFile.m are you using mass = `equilibrium` or defining the mass with an input value? There are times where the buoyancy calculated from your BEM solver may not be exactly equal to your theoretical calculations. Especially with low mesh resolutions where you may not be capturing the curvature of the body correctly. If you are using an input value it might be just a little more than the buoyancy WEC-Sim is told your WEC has. 
- If these do not help troubleshoot your problem, can you provide any parts of your model that we can review or even run ourselves? 

Hopefully these responses and actions help you troubleshoot your model, but let me know if you have further questions or need clarification. 

Nathan
National Renewable Energy Laboratory

nathanmtom, 2022-03-09:
@eleftheria-pap We have not heard from you in about two weeks so we are going to assume you have resolved your problem and close this issue. If you have further questions please reopen this issue. 

Cheers, 
Nathan

closed 2022-03-09 by nathanmtom