WEC-Sim/WEC-Sim #811: [BUG]Paraview_Visualization\RM3_MoorDyn_Viz

Hi-gone, 2022-02-24:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**

I encountered the same error in #674, then I tried to download the code in #682 , but the problem was still there 

**WEC-Sim file(s)**
WEC-Sim_Applications-master\WEC-Sim_Applications-master\Paraview_Visualization\RM3_MoorDyn_Viz

**To Reproduce**
Install the latest version of MoorDyn for WEC-Sim at: https://github.com/WEC-Sim/MoorDyn. Run BEMIO.
Run the applications case: Paraview_Visualization\RM3_MoorDyn_Viz
See error:

_State method of running WEC-Sim (command line)_

**Expected behavior**
I expect the case to run as is without issue

**Screenshots**

[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [Windows 10]
 - [MATLAB/Simulink] [ 2020b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v4.4]

**Additional context**
_Add any other context about the problem here._

akeeste, 2022-02-25:
Hi @Hi-gone 

Thanks for your question and using the template.

Unfortunately we have not been able to fix this bug as our team cannot replicate this issue consistently. Has this issue occurred frequently for you? I'd welcome any suggestions on how to cause this issue, as that would allow us to debug and find a way to solve this issue. 

I can take another look and search for a solution, but unfortunately it won't be robust unless there is a way to consistently cause this bug.

Adam

Hi-gone, 2022-02-26:
@akeeste Thank you for your reply
This issue occurs frequently.
I download the WEC-Sim on another computer.Install the MoorDyn for WEC-Sim.Run bemio.Run the applications case: Paraview_Visualization\RM3_MoorDyn_Viz. 
All the code is is downloaded from [https://github.com/WEC-Sim](url) on February 26, 2022.
 This error occurs on every run.

akeeste, 2022-03-01:
@Hi-gone 

Thanks for the update. I will search for another solution. If all else fails, we should be able to catch the error and repeat the commands to read the MoorDyn output files until they are fully written.

H0R5E, 2022-03-02:
Might be fixed by https://github.com/WEC-Sim/WEC-Sim_Applications/pull/9

akeeste, 2022-03-03:
@Hi-gone PR #814 should fix this issue. It should be merged into master next week.

Hi-gone, 2022-03-04:
@akeeste 
Thanks for the update.
My program is already running correctly.
Sincere thanks for your help.

closed 2022-03-10 by Hi-gone