WEC-Sim/WEC-Sim #815: [BUG] Read_AQWA interpretation of coordinate systems

hresing, 2022-03-04:
**Describe the bug**
We are using .LIS and .AH1 files from AQWA to get our .h5 file. To our understanding, this file should contain the information about the hydrodynamic body's center of mass and center of buoyancy, relative to the coordinate system in AQWA. However, when we run wecSim, the center of mass of our hydrodynamic body appears outside the body, and the free surface waterline is lower than prescribed in AQWA, as seen in the image below. Given the troubleshooting steps we have taken, it seems that this could be a bug in the way AQWA files are being read by bemio.
This theory is supported by the "noWaveCIC" case, where our geometry is initially oriented vertically as expected, and then rotates to an unexpected new equilibrium position. This is labeled in an image below.

**WEC-Sim file(s)**
Our best guess would be the Read_AQWA script used by bemio.

[2022 Flat wall hull updated.zip](https://github.com/WEC-Sim/WEC-Sim/files/8183000/2022.Flat.wall.hull.updated.zip)

**To Reproduce**
n/a
WecSim is being run via command line

**Expected behavior**
We would expect the center of mass and/or center of buoyancy to be located inside our geometry, at x=0, y=0, and below our expected waterline. 

**Screenshots**
[image]

[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB R2021b/Simulink Release2021b 
 - WEC-Sim v4.4

**Additional context**
We have been very deliberate about the export of stls from solidworks such that we are certain that the stl coordinate system is located at the waterline, and that the geometry input to AQWA has the same coordinate system. Everything is in meters. The mass properties used form solidworks are relative to this same coordinate system. 
Our desired output is to be able to constrain the body to rotate only in pitch, about a specific rotational axis, much like an OSWEC.

akeeste, 2022-03-09:
Hi @hresing 

Thanks for opening an issue and using our template. I'll do my best to answer your question.

It seems the core of this issue is the stl file. The stl file should not have it's origin at the SWL, but [at the body cg](https://wec-sim.github.io/WEC-Sim/master/user/workflow.html?highlight=stl#geometry-file). This is why your cg appears outside of the body, because the STL is defining the body's mesh that way. If we view the STL and the explorer at t=0, we see the same cg relative to the body mesh:
[image]
[image]

Is your cg supposed to be at a -x position, where the visualization is appearing? If so we can investigate Aqwa. Currently I see that ``body.cg=[0 0 -0.1160]``. 

Adam

akeeste, 2022-03-16:
Hi @hresing 

I'm checking in on this issue. Has the above response clarified the problem that is occurring? Please let me know if you have additional questions.

Adam

hresing, 2022-03-16:
Hi,
Thanks so much for the support - that lead us in the right direction, and we have figured out that when exporting our solidworks to stl, we have to choose the option of not moving our body to positive space.

akeeste, 2022-03-17:
Hi @hresing 

I am glad that this helped solve your issue. It sounds like your problem is solved so I will close this thread. If something else on this topic comes up, feel free to open a new issue .

Adam

closed 2022-03-17 by akeeste