WEC-Sim/WEC-Sim #824: [BUG] Multiple cable blocks in SIMULINK

ckberinger, 2022-03-13:
**Describe the bug**
The cable block in Simulink currently only allows for one cable based on how it is written.
You see in yellow, it is asking for a cable body mass using "cable(2).bodyMass". But this calls my cable matrix row 2 which has nothing in it because when I make cable(1), cable(2), cable(3), cable(4) it is all in row one and spans the columns. 
[image]

**WEC-Sim file(s)**
Here is a zip of my file. I am trying to use the cable blocks as 4 taut mooring lines. Any other advice on my setup is appreciated.
[6DOF.zip](https://github.com/WEC-Sim/WEC-Sim/files/8238730/6DOF.zip)

**To Reproduce**
Run the WEC-Sim file provided from the command window.

**Expected behavior**
I get this error when trying to run with multiple cables. But it is only an error for cable 2 through 4 and not 1. 
Error using wecSim (line 46)
Error due to multiple causes.
Caused by:
    Error using wecSim (line 46)
    Error in 'LUPA14sim/Cable2': Failed to evaluate mask initialization commands.
        Error using wecSim (line 46)
        Invalid setting in 'LUPA14sim/Cable2/Follower Drag Body/Body Mass' for parameter 'Value'.
            Error using wecSim (line 46)
            Error evaluating parameter 'Value' in 'LUPA14sim/Cable2/Follower Drag Body/Body Mass'
                Error using wecSim (line 46)
                Index exceeds the number of array elements. Index must not exceed 1.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release] 2021b
 - [WEC-Sim Release] V4.4

**Additional context**
I tried to make a new cable block or edit the current one but I could not figure out how to do that. I made a new cable slx file, broke the library link, and tried editing but it always reverted back. I looked at how the body class works and it does not use "body(2).mass" it uses "body.mass" even for the second body which makes me think it is a typo in the cable block. 
How can I edit the simulink blocks that reference the library so I can continue debugging? 

Thank you,
Courtney

dforbush2, 2022-03-15:
Hi @ckberinger 

I am taking a look at this issue today. You are correct that this seems like a mask initialization issue on the cable block, but I am confused by your model. Based on your wecSimInputFile.m, I would expect to see 4 distinct cable blocks and 5 constraint blocks in your *slx model, but there was only 1 constraint and no cables in the model that was included. I am also a bit confused geometrically: specifying the cable endpoints that you have, these points are contained in a single XY plane at Z = 0.991 m of depth: i.e., there is no vertical component to these connections, and they do not seem to connect dynamic bodies? This is a novel use case and you may consider, if the system is pretensioned and taut, if the physics might be closely approximated with the much simpler mooring matrix approach. 

But to the point of the bug: it seems possible this is a version issue. Previously, data types for 'set_param' blocks were finnicky when they appeared in mask initialization blocks, which is why the 'constant' parameter is set via different syntax than the variable names. However, as you note, this doesn't properly recognize the case of multiple cables.  I will push a pull request later after I have confirmed that this fix is reverse-compatible with older matlab versions, but to get you rolling immediately:

1) Right click on the cable block in the slx model, and select "Library Link >> Disable Library Link". (If this is greyed out, you may need to unlock the wecSimLib.slx library first. This can depend on how the block was imported into the model, but https://www.mathworks.com/help/simulink/ug/disable-links-to-library-blocks.html should get you started).

2) Now you can edit the mask function by right clicking block  "Mask >> Edit Mask".

3) Click the Initialization Tab

4) Find the lines under Follower drag body
And replace:
set_param(h,'Value', ['cable(' num2str(cable.cableNum) ').bodyMass'])
with:
set_param(h,'Value', num2str(cable.bodyMass))

5) Repeat 4) lines under base drag body

dforbush2, 2022-03-15:
Addressed by PR #827

dforbush2, 2022-03-15:
Hi @ckberinger  

I'm attaching a working example case for your geometry with multiple cables (thanks to @kmruehl for assistance finalizing this). This is not intended to accurately represent your real device as it is intended to operate, just to demonstrate modeling practices and the success of the bug fix.

Generally, note that:
* if an object (i.e., pto, cable, mooring, body, etc) is initialized in the wecSimInputFile, it should exist as a block in the slx file or its effect on dynamics will not be modeled, if the model even runs without error.
* cables couple two constraints and/or PTOs, which are then themselves typically connected to body blocks or reference frames.
[6DOF_updated.zip](https://github.com/WEC-Sim/WEC-Sim/files/8257214/6DOF_updated.zip)

closed 2022-03-15 by dforbush2

ckberinger, 2022-03-16:
Hi @dforbush2,

Thank you for fixing the cable block and explaining how to unlink the library.
I uploaded the wrong ZIP file, the confusion is completely my fault there. 
Here is the correct one, which has 4 cable blocks and 5 constraints. For context for the thread, this is a tank scale model which is moored to flume walls, hence the lack of Z change in the cable blocks. 
[CableBlockMooring.zip](https://github.com/WEC-Sim/WEC-Sim/files/8258067/CableBlockMooring.zip)

dforbush2, 2022-03-16:
Ah ok. This model looks a lot better. With the bug fix, this should run, but I do not quite think it is modeled as intended. 

All cables connect to a 6 DOF block. This allows one connected body to move in 6 DOF relative to the other connected entity: I think that this will allow the body to move in 6 DOF relative to the connected cable (i.e., it will act like the cable isn't really connected). I think this should also be a spherical constraint, also. The body still should be free to move in 6 DOF to the extent allowed by the cables (if they are stiff and pre-tensioned, this won't be very much). 

Generally, these cables are acting as a mooring:
Both mooring blocks are designed to couple one moving body to a fixed reference frame. There are 2 different included ways to handle this: mooringMatrix and moorDyn.

The cable block was actually originally intended to couple two moving bodies: we were seeing a lot of 2-body point absorbers coupled by cables that we did not have a good ready-made solution to model. It is essentially a logic check wrapped around a mooring matrix: “is it slack? If not, apply tension forces. If yes, apply zero force”. There is some limited functionality to include fluid drag effects on the cable. But for your application, which is effectively using it for a mooring, all of this is pretty low-order compared to moorDyn. 

So the question on mooring modeling approach becomes: do you expect the cable to go slack during sea states of interest? If not, use the mooring matrix. If so, using moorDyn should be plan A, with the cable implementation being a strong plan B. 

There is actually a 4th option too: using a multi-body linkage to model a perfectly rigid cable.  For your case, prescribing fixed constraints to four distinct points, this could crash the solver, but could be worth a try. Basically, you model the cable as two inextensible rods with a rotational/spherical constraint in the middle. In this way, when straight, the rod applies coupling force between connected bodies. But when “slack”, the rotational constraint moves the rods “out of the way” so doesn’t transmit force between the connected bodies. This avoids the logic check in the cable blocks, so it can run faster.