WEC-Sim/WEC-Sim #836: CWR in regular waves versus irregular waves

jwills801, 2022-03-20:
**Is your question request related to a problem? Please describe.**
I am trying to simulate an OSWEC in both regular and irregular waves using the same PI controller. I am using a PM spectrum with waves.H = 2.5 meters and waves.T = 8 seconds. I use the same period and wave height when I simulate regular waves. The PI gains were tuned for the regular wave case using the optimization toolbox in matlab - specifically fminsearch - to maximize energy absorption. These gains were implemented using the pto(1).c and pto(1).k parameters in wecsim.

I am trying to compare the capture width ratio obtained in the regular wave case versus the irregular wave case. I am getting the wave energy resource from waves.Pw, and I am using 18 meters as the width of the WEC. I calculate the absorbed power by multiplying the PTO force by the pitch velocity. I calculate the capture width by dividing the average absorbed power by waves.Pw. Then, to find the capture width ratio, I divide the capture width by the device width.

I am getting a CWR of 76% for regular waves and 87% for irregular waves. I am also getting that the wave resource is 4 times larger in regular waves than irregular waves.

**Describe the theory or implementation approach that you have a question**
I am confused by two things. 
     First, why is there so much less energy in irregular waves for the same significant wave height and period? Is this the expected result or am I doing something wrong?
     Second, why would the CWR be larger in the irregular wave case? I tuned the PI gains for the regular wave case so I thought that it would preform best at this specific frequency, and worse at other frequencies. I thought that in irregular waves, some of the energy would be in frequencies other than the one I optimized for, and my PI controller could be less effective at absorbing the energy in these frequencies. I figured that this suboptimal behavior at other frequencies would result in a smaller capture width ratio - which I have taken as a kind of stand in for efficiency.

**Personal project context**
I am still trying to figure out what is going on in WEC-sim and I have some confusions I need help in sorting out. I am researching wave energy in school.

**Describe the type of conclusion or resolution you are looking for**
I am hoping to achieve more understanding. If these results make sense, I would like to understand the rational. If these results do not make sense, I would like to figure out what I m doing wrong in my implementation.

**Additional context**
N/A

nathanmtom, 2022-03-20:
@jwills801 Thank you for your issue and I will do my best to answer your questions. 

To your first question, the `waves.T` and `waves.H` are defined differently for a regular and an irregular wave. In regular waves the OSWEC will encounter a single cosine wave with a wave amplitude of `waves.H/2`. While for irregular waves, the OSWEC will encounter multiple cosine waves that have energy within the [wave spectrum](https://wikiwaves.org/Ocean-Wave_Spectra). For irregular waves, the `waves.H` value is the significant wave height which is defined as the average of the top one-third wave heights measured in the wave elevation record. Therefore, the wave heights will on average be smaller than `waves.H` value defined in the wave input file. As such since wave power is proportional to the wave amplitude squared, if your average wave amplitude is halved the power of the waves is only a quarter which is consistent with your description above. If you want more details on the `waves.Pw` variable you can look within the `waveClass` where this is calculated for each wave environment.

To your second question, yes it is possible since the capture efficiency depends on the incident wave frequency and given the wave spectrum spreads energy across a range of wave frequencies the OSWEC may actually be encounter wave frequencies that have a larger capture width ratio than for the `waves.T` frequency you define for the wave spectrum. However, this can vary fairly dramatically depending on the mode of oscillation of the wave energy converter. A suggestion is that you can plot the capture efficiency from your regular wave sweep, when optimizing your PI gains, and see if you locally may have a strong growth in the capture width efficiency to either side of the `waves.T` value. Another item to check is how long of a time horizon are you using to calculate your average power. In regular waves, after about 40*waves.T of time the OSWEC should be in steady state and if you average across any number of integer wave periods your time average power should not vary. However, under irregular waves you will probably need to run between 100-150*waves.T of time before you have better time average power estimate given the random wave train. 

I hope these answers help guide you in your own research, but please let me know if you have other questions or need clarification on any of the above.

Cheers, 
Nathan 
National Renewable Energy Laboratory

jwills801, 2022-03-21:
Hello Nathan,

Thank you very much for your reply. I now understand that the difference in energy between regular and irregular waves is due to the way that the significant wave height is defined in each case. I had thought that keeping waves.H the same would be a fair comparison, but now I think that keeping waves.Pw the same in each case would result in a more fair comparison between irregular and regular waves. Is this reasonable?

As for the second part. I had already been lengthening the simulation time until the simulation time no longer effected the CWR for irregular waves. I have also been excluding the ramp up time while calculating averaging absorbed power. I did what you suggested and did a frequency sweep using regular waves of the same height using the same PI gains. So at each frequency, the only thing I changed was the frequency. The PI gains were still the ones optimized for the peak period of 8 seconds. I saw that the capture width ratio decreased as I moved away from the frequency at which the PI gains were optimized for.

[image]

This is the result I expected. The CWR decreases as the frequency moves farther from the frequency for which the PI gains were tuned. But, I am still confused why the irregular wave case has a larger CWR than the regular wave case when both cases use a PI controller tuned for the regular wave frequency.

nathanmtom, 2022-03-22:
Thanks @jwills801 for the follow-up to my responses, I'm glad to see that they seem to have helped answer some of your questions. 

To you first comment, by keeping `waves.Pw` constant between regular and irregular waves only keeps the energy within the wave system the same, but the OSWEC is being oscillated differently so it will still depend on what you are trying to compare.

To your second comment, appreciate you running the regular wave sweep to help understand your OSWEC model. Given the plot you have shown I would agree with you that moving to irregular waves that the capture width would have been reduced since the energy is spread across a wider range of frequencies. My next thought is have you checked the amplitude of your rotational oscillation is correct, or rather, not running up against any rotational constraints? If you are tuning the PI gains to assist the OSWEC in oscillating in resonance your amplitude of motion resulting from a `waves.H = 2.5 m` and `waves.T = 8 s` might be causing issues. The OSWEC cannot rotate more than 90 degrees positive or negative as it would hit the sea bed. In order to have a larger capture width in regular waves, you may need a larger amplitude of motion that you can simulate and by reducing the amplitude of motion you reduce your max power and capture width. A better check may be to reduce the wave amplitude to say 0.1 m, or smaller, and repeat your analysis. Given the much lower wave amplitude the OSWEC is less likely to run up against any motion limitations where you might see a larger capture width from regular waves than irregular waves. 

I hope these suggestions help you continue to debug your model. 

Cheers, 
Nathan
National Renewable Energy Laboratory

jwills801, 2022-03-23:
@nathanmtom Thank you so much for helping me through this. This has been great. I overlaid the irregular wave spectrum onto the CWR curve I made.

[image]

As for files, [here](https://github.com/jwills801/Li_research/tree/main/PI_frequencySweep/OSWEC) is the folder I've been using. I have just been changing the parameters of the same wecSimInputFile to change between the regular and irregular wave case.

nathanmtom, 2022-03-24:
Thanks @jwills801 this just helps confirm there wasn't an issue in the construction and implementation of the wave spectrum. 

Also appreciate you sharing your working directory to understand if you have any other model additions that could be affecting results. After reviewing there is nothing that is jumping out to me as incorrect or needs to be modified, so I am inclined to trust WEC-Sim's results. 

There are two additional items that I'd like you to try for me: 

1. Run with 3-5 other `waves.phaseSeed` values. Given WEC-Sim randomizes the phase angle for the individual wave components, there is the potential for having a higher than expected number of waves. I would expect that there to be some scatter in the mean power, but the question is what is the mean of these values. I would hope this might be closer, or even lower, than the capture width ratio for regular waves. 
2. Run your irregular wave run with `waves.freqDisc = 'traditional'` the default option is the equal energy method which reduces the number of wave components for faster simulation; however, the `waves.Pw` is calculated from the 'traditional' approach and perhaps there are some slight differences there than can account for the discrepancies you are seeing. 

Cheers, 
Nathan
National Renewable Energy Laboratory

jwills801, 2022-03-25:
Thank you again @nathanmtom. I really appreciate your help.

I tried tried using other phaseSeed values, and ran some cases without specifying a phaseSeed. The results did not change much. The irregular waves still had a higher capture width ratio. The same thing happened when I ran the irregular waves with the freqDisc set to 'Traditional' - the CWR was still larger in the irregular wave case.

I had noticed that the OSWEC is run in shallow waves, so waves.Pw is calculated using a much more complicated expression. I tried calculating the capture width ratio using the deep water expression for Pw and finally got a larger CWR for the regular wave case (around 1.5) than the irregular wave case (around 1). 

Do you think this finding is at all significant? Where can I find a derivation for the shallow wave power resource expression?

Thank you!

nathanmtom, 2022-03-25:
@jwills801 Thanks for following up with my request although they did not identify any issues, but seemed to confirm your previous results. 

In response to `waves.Pw` calculation, you are correct that the OSWEC is in shallow water so the deep water expression does not apply in this case. I am not sure what expressions you used, but deep-water regular and irregular have different equations for their calculation. Therefore, the calculation using the `waves.Pw` (which you can find within the wave class) should be choosing the right calculation method. If you want the full expressions for regular and how to transform into irregular I will point you to [this reference](https://ieeexplore.ieee.org/document/8114192).

After discussing with other members of the WEC-Sim team, we are in generally still in agreement with you that we would expect the capture width ratio to be reduced in irregular waves, but given the power at each wave frequency and the power within each wave (which is comprised of multiple waves) that there is still potential for a larger capture width as you are seeing. For example, I'm showing a plot below that includes the normalized wave spectrum and the wave power at each frequency in the wave. As you can see the frequency at which wave power is maximum is slightly longer than the peak wave period.

One other check you might want to complete, if not already, is to calculate the capture width ratio for other peak periods and plot on top of your last plot. Then we can see if at every period (Regular vs Irregular) your relative capture width is higher.

[image]

%% MATLAB code to reproduce the plot above %%
powerSpectrum = (1/2)*simu.rho*simu.g.*waves.S.*waves.dw.*sqrt(simu.g./waves.k.*tanh(waves.k.*waves.waterDepth)).*(1 + 2.*waves.k.*waves.waterDepth./sinh(2.*waves.k.*waves.waterDepth));
powerWaves = sum(powerSpectrum)
%%
close all
plot(2*pi./waves.w,waves.S/max(waves.S),...
     2*pi./waves.w,powerSpectrum/max(powerSpectrum),'--');xlim([0 16])
legend('Wave Spectrum','Wave Power');grid on
ylabel('Normalized Values');xlabel('Wave Period, T, [s]')
%%

jwills801, 2022-03-26:
Hello @nathanmtom. Thank you again for your responses. The reference you sent was especially helpful. I swept over frequencies for irregular waves while keeping the PI gains the same as they have always been. I set freqDisk to 'Traditional'.

[image]

nathanmtom, 2022-03-28:
@jwills801 Thanks for completing the additional sweep. The results appear to indicate that until you move sufficiently far away from the target resonance period (of 8 s) you still get a large response which seems to continue to elevate the power production. 

I would like to confirm that the optimum spring and damping PTO coefficients are indeed producing the maximum power at the target wave period of 8 s? When I review you repository you appear to have a script to assist in determining this values, but have you double checked the procedure using a traditional two-dimensional search of iterating across a wide spread of damping and spring coefficient pairs? My thought is that potentially the true optimum in regular waves is higher than you have shown above such that it could be increased to exceed the irregular wave case.

Also if possible, can you include a plot of the irregular time history of pitch response for the flap and the corresponding fast-Fourier transform? I'd like to double check that you are indeed getting the anticipated response for the irregular waves.

jwills801, 2022-03-29:
Hello @nathanmtom,

I'm sorry. The PI gains are not what I said they were. I said I was using the gains from a two-dimensional search, but I was actually using gains found by linearizing the dynamics. The code I use to do the 2D search is [here](https://github.com/jwills801/Li_research/tree/main/Find_optimal_PI_gains). The gains are a little different and the CWRs are a little different but the pattern still holds. Here is the plot I made earlier, but using these new gains. 

[image]

The conclusion from these results seem to me to be the same as before. Here is the time history of the pitch angle of the flap, and below that is the fft of that. For the lower two plots I used a peak period of 8 second. I was surprised that the fft seems to be shifted further towards higher periods than the PM spectrum.

[image]

`figure, plot(simu.time,output.bodies(1).position(:,5)), ylabel('Pitch Angle (rad)'), xlabel('Time (s)'), xlim([1000 1500])`

[image]

`X = output.bodies(1).position(1:end-1,5);` 
`T = simu.dt;                      % Sampling period`       
`Fs = 1/T;                         % Sampling frequency`                    

`L = length(simu.time(1:end-1));   % Length of signal`

`Y = fft(X);`
`P2 = abs(Y/L);`
`P1 = P2(1:L/2+1);`
`P1(2:end-1) = 2*P1(2:end-1);`
`f = Fs*(0:(L/2))/L;`

`figure, yyaxis left, plot(1./f,P1), xlim([2 14]), xlabel('Period (s)'), ylabel('|P1(f)|')`
`yyaxis right, plot(2*pi./waves.w,waves.S), ylabel('Spectrum (m^2 s/rad)')`
`title('Single-Sided Amplitude Spectrum of Pitch Angle')`

nathanmtom, 2022-03-30:
@jwills801 Thank you for the clarification about how you obtained your optimum PI search. I looked at the files you linked for the PI gain search and it appears you used the `fminsearch` function within MATLAB which I have minimal experience. If this function is completing the search for you, have you confirmed from a 2D vector search you are indeed obtaining the optimum PI gains? 

Thank you for the additional time history and corresponding FFT as this helps me see that the pitch response seems consistent with the incident wave spectrum. The oscillation amplitude also varies with frequency so it is possible to have larger amplitudes of response off of your peak frequency depending on the hydrodynamics. 

At this point, I do not seem to see any issues with the input files and the output results from the WEC-Sim model. Unfortunately, trying to dive deeper into your files to help troubleshoot the question you have about capture width is starting to move outside of the scope of this issues board. Although you still may have questions about your specific results, I do not see any reason not to trust the WEC-Sim outputs and so I would say there is not an issue with the code itself. 

Perhaps if you were to implement other system identification processes, or continue on with your work, you may either resolve your current issue or find additional data that can help us track down an issue with the code.

Cheers, 
Nathan
National Renewable Energy Laboratory

jwills801, 2022-03-30:
Hello @nathanmtom, I really appreciate all the time you've put into helping me. I do have another question though. I was looking through the paper you sent me which has a lot of the equations which wecSim uses. I noticed that when computing the wave energy resource for regular waves in shallow water (line 594 of waveClass.m) the code is

`obj.Pw = rho*g*(obj.A).^(2)/4*sqrt(g./obj.k.*tanh(obj.k.*obj.waterDepth))*(1+2*obj.k.*obj.waterDepth./sinh(obj.k.*obj.waterDepth));`

But, from the paper - and from line 651 where the wave energy resource for irregular waves in shallow water is calculated, I think line 594 should instead be 

`obj.Pw = rho*g*(obj.A).^(2)/4*sqrt(g./obj.k.*tanh(obj.k.*obj.waterDepth))*(1+2*obj.k.*obj.waterDepth./sinh(2*obj.k.*obj.waterDepth));`

The only change is the factor of 2 inside of the `sinh` function.

When I change this line in my local instance of waveClass.m, I get the expected result of having a larger capture width ratio for regular waves than for irregular waves.

Do you also think this line should be changed?

nathanmtom, 2022-03-30:
@jwills801 Thank you for following up with this point and double checking our wave power calculation within WEC-Sim. I stand corrected and there is a bug in the wave power calculation. You are correct that the full wave power expression for regular waves is missing a factor of 2 within the `sinh` function which is significant since that function grows very quickly. 

I am going to submit a bug fix to the `master` branch of WEC-Sim to correct the bug.

jwills801, 2022-03-30:
Hello @nathanmtom, yes! Thank you so much for all your help. I have no more questions about this. Thank you again!

nathanmtom, 2022-03-30:
Thanks @jwills801, I've pushed the changes to the WEC-Sim master branch but have documented this change in our `dev` branch with PR #841 . This way you will get credit for brining this issue to our attention. Thank you for doubling checking our wave power expression