WEC-Sim/WEC-Sim #837: Constraint Angle[BUG]

FFxiaolang, 2022-03-21:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
_A clear and concise description of what the perceived bug is._
I was interested in issue 48 and tried to run it, but I encountered the following error .
Error evaluating parameter 'TranslationCartesianOffset' in 'AH1/Rotational2/Rigid Transform (Position)1'
So how can I change the constraint angle? 
[image]

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to '...'_
2. _Click on '....'_
3. _Scroll down to '....'_
4. _See error_

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

MATLAB/Simulink 2020b
WEC-Sim 4.4
**Additional context**
_Add any other context about the problem here._

FFxiaolang, 2022-03-21:
[QU.zip](https://github.com/WEC-Sim/WEC-Sim/files/8313576/QU.zip)

FFxiaolang, 2022-03-21:
constraint(1).initDisp=deg2rad(-45); 
It has an error.

FFxiaolang, 2022-03-21:
https://github.com/WEC-Sim/WEC-Sim/issues/48

akeeste, 2022-03-22:
Hi @FFxiaolang 

Thanks for opening an issue, I will try to address your question.

``constraint(1).initDisp=deg2rad(-45);`` gives the error message ``Error evaluating parameter 'TranslationCartesianOffset' in 'AH1/Rotational2/Rigid Transform (Position)1'`` because initDisp is meant to be a structure. Simulink tries to dot index into this structure to get the variable ``constraint(1).initDisp.initLinDisp``, but only gets a float ``deg2rad(-45)``. ``initLinDisp`` has to be a [1x3] vector including the initial linear displacement in x, y, and z. Instead use: ``constraint(1).initDisp.initLinDisp = [0 deg2rad(-45) 0]``

Secondly, the WEC-Sim constraintClass and ptoClass don't use initial angles like the bodyClass does. If you want to change the angle of the constraint or PTO axes, then set ``constraint.orientation.z`` and ``constraint.orientation.y`` [For translational PTOs](https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#modifying-constraints-and-ptos), ``z`` defines the axis of linear motion, and for rotational PTOs ``y`` defines the axis of rotation. Note that changing the orientation of the fixed constraints (1, 3) will not affect the simulation. You will only see a difference in how the output constraint force is divided into its various components. Similarly, changing ``constraint.orientation.z`` for the rotational constraints will only affect the angle that corresponds to zero in the output data

I hope this helps clarify. Let me know if you have any more questions.
Adam

FFxiaolang, 2022-03-23:
@akeeste  Hi! I set pto(1).orientation.z=[0,0,deg2rad(-45)]. But code has no effect.  I want to change the pto  angle of z , such as this figure.
[image]

FFxiaolang, 2022-03-23:
[image]

FFxiaolang, 2022-04-02:
@akeeste  Thank you so much for all your help. I have no more questions about this.

closed 2022-04-04 by akeeste