WEC-Sim/WEC-Sim #843: A problem of running the example :$WEC-Sim\tutorials\RM3----(matlab use)

woosalute, 2022-04-09:
i am a new to WEC-Sim, i am trying to run the bemio.m in the tutorial document (E:\WEC-Sim\tutorials\RM3), but the matlab has always shown the error below:
                            The function or variable 'Read_WAMIT' is not recognized.
what can i do to solve it? i would  appreciate if  you can answer my question.

woosalute, 2022-04-09:
[image]
this is my matlab interface

nathanmtom, 2022-04-11:
@woosalute Thank you for submitting your issue; however, we require users to submit their issues following one our templates based on your inquiry. Please submit a new issue following the template or update this existing issue following said template. I've included some screen shots below to help walk you though this process.

[image]

[image]

woosalute, 2022-04-13:
thank you for your reply! after two days working ,i have solved the problem that i have had,but i still thank you very much

closed 2022-04-13 by nathanmtom