WEC-Sim/WEC-Sim #844: Question about powerInternalMechanics

aristi-pap, 2022-04-11:
**Is your question request related to a problem? Please describe.**
Hello! I have a problem with plotting power and more precisely why I get negative values, while I was expecting positive ones. 

**Describe the theory or implementation approach that you have a question**
From the WecSim manual it is understood that the power equation is given by
[image]
If the Κpto is equal to zero, as in my case, the equation is Ppto=Cpto*velocity^2, which should result in positive values (Cpto>0).
From all this I conclude that  output.ptos.powerInternalMechanics = - output.ptos.forceInternalMechanics.*output.ptos.velocity

**Personal project context**
In my model I use Linear PTOs (translational), PTO Stiffness is equal to zero and the WECs move only in heave mode. To plot the power of the first WEC I use the command  plot(output.ptos(1).time(1501:1801),output.ptos(1).powerInternalMechanics(1501:1801,3)) .
The figure that comes as a result is the following one: 
[image]
But, whenever I solve for Power using the equation P =-F * velocity or P=Cpto*velocity^2 I get positive power and this figure.
[image]
where F = output.ptos.forceInternalMechanics(1501:1801,3) and v = output.ptos(1).velocity(1501:1801,3).
As you can see I get the same values but opposite signs.

**Describe the type of conclusion or resolution you are looking for**
It would be very helpful if you could explain to me why this problem occurs and if I do not understand something about how the power is defined.

**Additional context**
N/A

nathanmtom, 2022-04-11:
@aristi-pap Thank you for your question and I will do my best answer your question. 

Your issue arises from sign notation as within WEC-Sim the F_{PTO} value is being as you have written above, but for power WEC-Sim does not add a negative sign in front of the F_{PTO} value. The easiest is to look at how the calculation is being handled within the PTO block below. Therefore, WEC-Sim uses the notation that negative power means power is being taken out of the system, while positive power means power is being sent into the system. The easiest correction for your calculations to align with WEC-Sim is to remove the - sign in front of your power calculation.

[image]

I hope this answers your issue, but let us know if you need more clarification.

nathanmtom, 2022-04-20:
@aristi-pap Did my explanation above help resolve your question about WEC-Sim notation? If we do not hear from you in the next few days we will close the issue.

aristi-pap, 2022-04-21:
@nathanmtom Yes, it helped me. Thank you for your help!

closed 2022-04-21 by nathanmtom