WEC-Sim/WEC-Sim #848: [Rotational PTO Actuation Torque]

Hi-gone, 2022-04-16:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._

**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

akeeste, 2022-04-19:
Hi @Hi-gone

Thank you for opening an issue and using our template.

You can definitely use the ``Rotational PTO Actuation Torque`` block with a floating base. In this case the PTO block's torque and response are both of the Follower port (`F`) relative to the Base port (`B`). Typically the follower is the flap and the Simscape base is the OSWEC base. See the ports labeled on the block here:
[image]

You can likely use the same hydraulic PTO-Sim model that is in this Application case. However it might need some adjustments to account for a different physical set-up and the fact that the base can also move.

Please let me know if this addresses your issue and if you have any more questions.
Adam

Hi-gone, 2022-04-23:
@akeeste 
Thank you for your explanation,I guess the input in Rotational PTO Actuation Torque is relative speed.For two bodies in relative motion,they should be subjected to opposite torques.So is this output reassigned in the next segment?

akeeste, 2022-04-25:
@Hi-gone 

The ``Rotational PTO Actuation Torque`` block has a relative torque input, and the ``Rotational PTO Actuation Motion`` block has a relative motion input. Even though these blocks take relative speed and torque, Simscape will automatically apply the correct convention to each body so that each has an opposite and opposing force on it.

Adam

akeeste, 2022-05-13:
Hi @Hi-gone 

Do you have any more questions on this issue? If not I will close out this issue next week. Feel free to open another for any more questions you have.

Thanks,
Adam

closed 2022-06-01 by nathanmtom