WEC-Sim/WEC-Sim #849: [WEC-Sim Applications] Applying the Generator Damping to the Rotary Generator

wectechpol, 2022-04-19:
**Note: N/A

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications/PTO-Sim/RM3/RM3_Hydraulic_PTO) 
_Please identify the WEC-Sim applications case in question._

Hello, 

First of all, I would like to thank everyone who has contributed to this open-source. I'm currently working on optimizing RM3 with HPTO and implementing it in a selected location. I'm affiliating the B_PTO (assuming that it is a spring-damper system) and according to the optimal B_PTO value, I'm planning to tune the hydraulic parameters.

Motor displacement and generator damping are some of the significant parameters to improve the performance.

Since the generator torque is calculated with the next equation: genTorque = (delPressmotorVol/(desiredSpeed1.05))*angVel
Where 1.05 accounts for the mechanical efficiency and the term angVel/desiredSpeed accounts for the volumetric efficiency.

On the other hand, according to angVel * genaratordampingcoefficient can also give the generator torque. 

Do you have any suggestions on where to implement the generator damping in the Simulink file? I have already added the value through ptoSimInputFile.m.

I also wonder how will I monitor the motor displacement times alpha outputs from variableMotorVolume.m

Hope I'm clear with my question and kindest Regards

**Workstation Information (this section MUST BE completed):**
 - OS: [Monterrey12.0.01]
 - [MATLAB/Simulink Release 2021a]
 - [WEC-Sim Release v4.1]

jleonqu, 2022-04-20:
Hello,

For the generator damping using PTO-Sim, you would need to change the input of the Hydraulic Motor block a little bit. If you go inside the block, this is the part where the equation of motion for the shaft is implemented:

[image]

There is a term associated with the hydraulic torque, the generator torque, and a friction torque. There is no damping term in the equation, but you can add it. You can add your own damping term inside the block, or you can add the term outside the block as part of the genTorque input:

[image]

There has been some work on improving the PTO-Sim library to include this damping term, among many other improvements. The new version will be available soon.

Regarding the motor displacement from variableMotorVolume.m, you can monitor that variable with the motorVol output in the Hydraulic Motor block.

wectechpol, 2022-04-22:
Hi Jorge,

Thanks for your explanation. I still have two questions. 

1- How is it mathematically possible to add the the damping term to the torque as an addition term?

2-variableMotorVolume.m is using an equation to adjust the the motoVol, is there any theoretical justification of that  equation, because unfortunately I didn't understand how it works and changes the motor displacement?

Thanks a lot in advance.

jleonqu, 2022-05-11:
Hi @wectechpol 

1. You the two options that I mentioned previously. For the first option, which is adding the damping term internally, it would be something like this:
[image]

2. This questions was answered in this [comment](https://github.com/WEC-Sim/WEC-Sim/issues/544#issuecomment-808327556), let me know if this explanation is enough or if you need further details.

jleonqu, 2022-06-01:
Hi @wectechpol 
Does this answer your question?

wectechpol, 2022-06-01:
Hi Jorge,
Thank you for your assistance, all good! 
Kindest Regards,
wectechpol

closed 2022-06-01 by wectechpol